prusa3d/PrusaSlicer #3912: printer resets with gcode from version 2.2.0

lcemoore, 2020-03-25:
### Version
2.2.0+win64

Win 10 Home
V. 1909
OS build 18363.720

Prusa MK3S 
FW 3.8.1-2869
Printer is built as originally designed- stock

### Behavior
The printer resets when printing with GCode created with the newest prusa slicer. 
This occurs for all printing attempts when printing with octoprint, SD card, and when using pronterface.

When printing using octoprint or the SD card, there are no error messages.
When printing with pronterface you get a bed preheat error instead of a reset

Slicing with Cura v4.4.0 works every time.

The reset or error occurs once the the model height reaches 1.0 - 1.2 mm.  This has been a consistent failure point for over 20 attempts to print using the prusa slicer. 

Additional information
I installed the prusa-slicer 2.2.0 RC5 from gethub the day before, which also failed in the same manner.
I also tried uninstalling and reinstalling the software with no effect. 

[test-cura.zip](https://github.com/prusa3d/PrusaSlicer/files/4380826/test-cura.zip)

rtyr, 2020-03-25:
300C might be too much for heatbed :-). Can I close this issue?
[image]

lcemoore, 2020-03-25:
Please leave this issue open until I can verify this in my installed software.  Additionally, I must point out that I did not make any changers to the installed filament profiles other than to create a custom profile for hatchbox filament, and have not yet made any adjustments to alter the copied profile from its default settings except for infill %.  Also I watched at least 10 prints from start to failure and did not observe any change to bed temperature being executed during the print leading up to failure.  I will add additional information later today,  Lets say in about 10 hours from the time of this reply

lcemoore, 2020-03-25:
Ok, you got me.  Now if I could just figure out why that value was at 300C.  Anyway with the value at 70C, all is well.  I  now officially merit the cone of shame.  Thanks for your help...

Close away.

closed 2020-03-26 by lukasmatena