prusa3d/PrusaSlicer #3913: Does not start (double click on prusaslicer.exe but nothing happen!)

Manuelpru, 2020-03-25:
### Version
**PrusaSliser 2.2.7**

### Operating system type + version
**Win 7 64 bit home premium SP1 - AMD Athlon II P320 Dual-core Processor, 4 GB ram, ATI Mobility Radeon HD 4200 Series**

### 3D printer brand / version + firmware version (if known)
**Prusa I3 MK2 original assembled (no kit) / Firmware 3.2.3 (lastone)**

### Behavior
**click on prusaslicer.exe and does not starts, nothing happen! The mouse arrow start to load something but then stop. I ve already had this problem some month ago when wanted to install V 2.2.9, solved installing V 2.2.8, all worked fine. Now, it's some week I do not print something, today I was going to launch the software and nothing happen, so I delete V 2.2.8 and intalled V 2.2.7 but it doen't works!**

**Win 7 event viewer says:**

Bucket errato 4161293701, tipo 366438298
Nome evento: APPCRASH
Risposta: Non disponibile
ID CAB: 0

Firma problema:
P1: prusa-slicer.exe
P2: 2.1.0.0
P3: 5d7f5393
P4: atio6axx.dll
P5: 6.14.10.9704
P6: 4b9e4b1c
P7: c000041d
P8: 0000000000d28ffb
P9: 
P10: 

File allegati:
C:\Users\Manuel\AppData\Local\Temp\WERC7D1.tmp.WERInternalMetadata.xml

I file potrebbero essere disponibili qui:
C:\Users\Manuel\AppData\Local\Microsoft\Windows\WER\ReportArchive\AppCrash_prusa-slicer.exe_1be5b28771f26310b2a1ba886174a03619880ac_13d3cc92

Simbolo analisi: 
Nuova ricerca di soluzioni: 0
ID rapporto: 7ce5c599-6ea5-11ea-93d5-00266c6ff818
Stato rapporto: 0
------------------------
Nome dell'applicazione che ha generato l'errore: prusa-slicer.exe, versione: 2.1.0.0, timestamp: 0x5d7f5393
Nome del modulo che ha generato l'errore: atio6axx.dll, versione: 6.14.10.9704, timestamp: 0x4b9e4b1c
Codice eccezione: 0xc000041d
Offset errore 0x0000000000d28ffb
ID processo che ha generato l'errore: 0x1550
Ora di avvio dell'applicazione che ha generato l'errore: 0x01d602b23c6ae55b
Percorso dell'applicazione che ha generato l'errore: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
Percorso del modulo che ha generato l'errore: C:\Windows\system32\atio6axx.dll
ID segnalazione: 7ce5c599-6ea5-11ea-93d5-00266c6ff818

**I already know probably you will write about video driver problems but my pc it's all ok...or it seems al least! I already delete and reinstalled video drivers, for now I don't know what to do, I am not a beginner but not even an expert...so please tell me how I can solve

thanks!**

bubnikv, 2020-03-25:
> I already know probably you will write about video driver problems

Exactly.

> Percorso del modulo che ha generato l'errore: C:\Windows\system32\atio6axx.dll

You are not the only one.

https://community.amd.com/thread/184943

bubnikv, 2020-03-25:
https://community.adobe.com/t5/photoshop/photoshop-cs6-crashes-atio6axx-dll-faulting/m-p/5155764

https://support.transmagic.com/hc/en-us/articles/206986713-Application-Crash-atio6axx-dll

Unfortunately I have no idea whether this helps.

Manuelpru, 2020-03-25:
I am so frustrated, my pc usually doesn't give me problems, ok it's a bit old but still my friend! :)

I have 2d-3d softwares that I use without problems...prusa does not want to work!

Thanks for links, I already knews it but I am a bit scared on touch that files...

bubnikv, 2021-09-22:
Likely a system installation issue. Closing.

closed 2021-09-22 by bubnikv