prusa3d/PrusaSlicer #3914: Installer reboots PC without waiting for user to respond to reboot prompt

poshcodebear, 2020-03-25:
### Version
2.2.0

### Operating system type + version
Windows 10 Home 1909 

### 3D printer brand / version + firmware version (if known)
Not applicable, issue during install

### Behavior
On finishing install, it prompts to reboot now or reboot later, but goes ahead and restarts without waiting for user input.

Steps taken:
- Download and install Drivers and Apps 2.3.0 from prusa3d.com and select only PrusaSlicer and Utilities from the custom menu

#### Project File (.3MF) where problem occurs
Not applicable, issue during install

poshcodebear, 2020-03-25:
I searched for this issue but didn't find it; I'll try to replicate it on another machine and see if it does the same thing and report back.

poshcodebear, 2020-03-25:
I was able to replicate this on another machine.

Other than changing the installation to Custom and deselecting MINI (the only other item selected by default for me) and unchecking the box to put a shortcut on my desktop, I just used the defaults.

lukasmatena, 2020-03-25:
If this happens with the installer for 2.2.0, it is quite unfortunate. The issue was reported in #3757 and #3198, which were closed as fixed. I'll let the people responsible for the installer know, maybe they'll ask for more data. Thanks for the report.

Erasmo12321, 2020-04-12:
Also had this issue when installing 2.2.0. Not too thrilled about it, had a multi-day transcode interrupted because of this.

Dschadu, 2020-04-13:
Had the same issue minutes ago with the installer for version 2.2.0.
It just restarts without waiting for a user input. 
OS: Win 10 Pro x64 Version 1903 (Build 18362.720)

LoyalNine1487, 2020-04-16:
Same issue here on Windows 10 Pro x64 1903 (Build 18362.778)

xy172, 2020-07-04:
My wife and I also had this issue today on both of our computers, it restarts before questioning whether you wish to restart or not rather than after selecting that option. (confirmed by cancelling the restart before the installer was closed)

sobieh, 2021-01-12:
Why wasn't it fixed in 2.3.0 ?

Janys, 2021-01-12:
> Why wasn't it fixed in 2.3.0 ?

Hello, I'm the person directly responsible for the installer. I'm very sorry for the trouble, we tested the installer internally and the issue did not appear, so we considered it fixed. I would like to ask you whether you'd be ok with sharing some more info with me, so I can further analyze the source of the issue.

Could you please tell me, which installer you downloaded? Was it the cca 360 MB package from prusa3d.com/drivers? If you run the installer again, is your PC automatically rebooted again? Could you please download the installer from this direct link and try installing it again? https://www.prusa3d.com/downloads/drivers/prusa3d_win_2_3_0.exe

sobieh, 2021-01-12:
I have found the installer log entry saying:
`2021-01-12 11:31:29.760   Run as: Current user
2021-01-12 11:31:29.760   Type: Exec
2021-01-12 11:31:29.761   Filename: C:\Users\<username>\AppData\Local\Temp\is-2FKEH.tmp\vc\vcredist_x64.exe
2021-01-12 11:31:29.761   Parameters: /install /quiet /norestart
2021-01-12 11:31:31.384   Process exit code: 0
2021-01-12 11:31:31.390   Need to restart Windows? Yes
2021-01-12 11:31:31.392   Deinitializing Setup.
2021-01-12 11:31:31.401   Restarting Windows.
2021-01-12 11:31:31.405   Log closed.`

This is caused by vcredist_x64.exe and seems to be known issue with /norestart not really working.
You should be able to reproduce this by uninstalling vcredist and running the prusa slicer installer again.

Janys, 2021-01-12:
> Yes i was using exactly this installer file.
> 
> I have found the installer log entry saying:
> 
> ```
> 2021-01-12 11:31:29.760   Type: Exec
> 2021-01-12 11:31:29.761   Filename: C:\Users\<username>\AppData\Local\Temp\is-2FKEH.tmp\vc\vcredist_x64.exe
> 2021-01-12 11:31:29.761   Parameters: /install /quiet /norestart
> 2021-01-12 11:31:31.384   Process exit code: 0
> 2021-01-12 11:31:31.390   Need to restart Windows? Yes
> 2021-01-12 11:31:31.392   Deinitializing Setup.
> 2021-01-12 11:31:31.401   Restarting Windows.
> 2021-01-12 11:31:31.405   Log closed.```
> 
> This is caused by vcredist_x64.exe and seems to be known issue with /norestart not really working.
> You should be able to reproduce this by uninstalling vcredist and running the prusa slicer installer again.
> ```

Looks like you're 100% right, thank you so much! I have removed the VCRedist installer from the package and I'm currently in the process of testing the new build.

poshcodebear, 2021-01-15:
> This is caused by vcredist_x64.exe and seems to be known issue with /norestart not really working.
> You should be able to reproduce this by uninstalling vcredist and running the prusa slicer installer again.

Good catch!

bubnikv, 2021-04-14:
@Janys updated the installer to not install vcredist, thus there will be no restart.
Starting with PrusaSlicer-2.3.1-rc, we are adding api-ms-win-crt-runtime-l1-1-0.dll to the PrusaSlicer installation, so there will be no need to install vcredist even if api-ms-win-crt-runtime-l1-1-0.dll is missing on the target system.

closed 2021-04-14 by bubnikv