prusa3d/PrusaSlicer #3915: Brim doesnt connect to part

Walderbash, 2020-03-25:
### Version
Prusaslicer 2.2.0_win64

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
Open the slicer, add any part.
I used 0.3mm Draft preset.
Set brim width to 2mm. Slicer preview shows brim disconnected from part. When printing the brim is also disconnected from the part. A brim always needs to be connected to be of any use. Problem does not occur in 2.1

_Is this a new feature request?_
This is a bug 

#### Project File (.3MF) where problem occurs
[image]
[image]
Example project:
[sb nub.zip](https://github.com/prusa3d/PrusaSlicer/files/4383331/sb.nub.zip)

lukasmatena, 2020-03-25:
This is correct. Elephant foot compensation shrinks the object to compensate for... well, the elephant foot. If there is a gap between the object and the brim, it is not set correctly and compensates more than the actual elephant foot would need.

Walderbash, 2020-03-25:
So the standard procedure how to setup a MK3S would be to tune the Z offset to produce a .2mm elephant foot out of the box, and then compensate for that in the slicer? That seems the wrong way round.

rtyr, 2020-03-25:
Looks like duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3870.

bubnikv, 2020-03-26:
> So the standard procedure how to setup a MK3S would be to tune the Z offset to produce a .2mm elephant foot out of the box, and then compensate for that in the slicer? That seems the wrong way round.

Elephant foot effect is inevitable if you squish the plastic sufficiently on the 1st layer to stick well. Your photos indicate to me that you likely want to lower the live-Z a little bit. Maybe you will not need the brim then as your object will stick better.

> I used 0.3mm Draft preset.

I thought it may have had some effect, but the 1st layer is still printed with 0.2mm layer height.

@rtyr would you please print this object?

rtyr, 2020-03-26:
Seems OK to me. 
Top - textured sheet
Bottom - smooth sheet
[image]

Walderbash, 2020-03-26:
Thanks for the quick reply, I did some tuning and found that when I lowered Live Z by 25 micron there was just enough squish to make the brim attach again. 

Normally I dont need a brim for good adhesion, but in this case I also had a high hollow tube, for me these tend to let go of the bed due to the small footprint:
[image]

malcinator, 2020-05-21:
Perhaps it could be possible to deactivate Elephant Foot Compensation when enabling the brim and vice-versa?

closed 2021-08-12 by lukasmatena