prusa3d/PrusaSlicer #3916: Crashes since adding custom bed texture and model

JohnOCFII, 2020-03-25:
### Version

PrusaSlicer-2.2.0-rc4+-202003151124

### Operating system type + version

Mac OS X 10.15.3 (19D76)

PrusaSlicer
Version:   2.2.0-rc4+
Build:     PrusaSlicer-2.2.0-rc4+-202003151124

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Catalina Version 10.15.3 (Build 19D76)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 ATI-3.5.5
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 555X OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix

### 3D printer brand / version + firmware version (if known)

RailCore 300ZL with Duet 2 Ethernet w/ 3.01-RC1 Firmware 

### Behavior

Since adding a custom bed texture and model, I'm seeing frequent PrusaSlicer crashes when switching printers or attempting to modify the bed shape or size.

Gist with full crash info is here: https://gist.github.com/JohnOCFII/6eb23931fccc264c9e7b764d0f7bbfb9

_Is this a new feature request?_

No
[face_shield_set_x4 copy.txt](https://github.com/prusa3d/PrusaSlicer/files/4383936/face_shield_set_x4.copy.txt)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

JohnOCFII, 2020-03-25:
Also adding the SVG and STL used: 
[Overlay_no_numbers.svg.zip](https://github.com/prusa3d/PrusaSlicer/files/4383950/Overlay_no_numbers.svg.zip)
[railcore.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4383953/railcore.stl.zip)

lukasmatena, 2020-03-25:
The backtrace reminded me of #3371. This seems to be the same issue in different place, I assume it will have the same solution. I simply copy-pasted the patch into the other function with 0054134.

@FidelCapo If you're able to reproduce the crash with 2.2.0, could you please retest in latest master to confirm the fix has helped? Thanks.

lukasmatena, 2020-03-26:
@JohnOCFII We did not manage to reproduce the problem on devices that we have available. If you can reproduce it consistently, would you mind testing the following build to check if it is fixed? Thank you very much.

https://drive.google.com/open?id=1fsz57uLdcryjvUP1mEXTKRJX9xevoWEZ

JohnOCFII, 2020-03-26:
Installing and testing now.  Thanks.  I can usually get it to crash within 3-4 minutes of usage.

JohnOCFII, 2020-03-26:
@lukasmatena - I've been unable to get PrusaSlicer to crash with the test version you posted above.  I've been in an out many times, changing bed textures, bed sizes, switching between RailCore and MK3, etc.  No crashes at all!  Good job!  Thanks.  I'll let you close the issue in case you need to add any tracking info, etc.

lukasmatena, 2020-03-27:
@JohnOCFII Thanks a lot for the testing. I'd consider this solved then.

closed 2020-03-27 by lukasmatena