prusa3d/PrusaSlicer #3917: PrusaSlicer isn't able to slice a stl.

rmayolp, 2020-03-26:
### Version
_Version of PrusaSlicer: 2.2.0+

### Operating system type + version
macOS Catalina 10.15.3

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+MMU2S / FIRMWARE 3.8.1 (1.0.6)

### Behavior
* PrusaSlicer does not work properly. It is no able to slice correctly the stl.
* I attach stl file and the slice result.
  
#### Project File (.3MF) where problem occurs
[DronsKIDScreuLogo1_3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4386693/DronsKIDScreuLogo1_3.3mf.zip)
[image]
[image]

foreachthing, 2020-03-26:
I just opened your project in PS: you really need to fix your model first:

[image]

rtyr, 2020-03-28:
The model needs to be fixed as @foreachthing mentioned. You can try "fix through the Netfabb" option in the PrusaSlicer.

closed 2026-09-29 by prusaslicer-bot[bot]