prusa3d/PrusaSlicer #3919: MMU2s and supports on prusaslicer 2.2.0 issue

palimbro, 2020-03-26:
### Version
2.2.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 64bits
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
MK2.5s with MMU2s

### Behavior
Problem with size of purge when activating the supports on MMU2s profiles, without supports it's ok

Not this problem before 2.2.0 version
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-03-26:
Please provide 3MF

palimbro, 2020-03-26:
Hello,
I will try to send you an other model, because this one is for my business.

Best regards

palimbro, 2020-03-26:
This is 2 3MF, one with supports and the issue, and the other without support working
[4C_Niko_Dog_mf with supports.zip](https://github.com/prusa3d/PrusaSlicer/files/4387297/4C_Niko_Dog_mf.with.supports.zip)

bubnikv, 2020-03-26:
Thanks for heads up. The issue is caused by
2fef16c39aea81831ac376caae472b51082d6582.
We are looking into a fix.

palimbro, 2020-03-26:
Do you think, is there an alternative to continue using this version of
PrusaSicer, because the pause and gcode insertion is very helpful for me?

Thanks for your support

bubnikv, 2020-03-26:
we will likely release a patch soon. This is a too serious bug.

palimbro, 2020-03-26:
Thank you ;)
Have nice day

bubnikv, 2020-03-27:
So the issue is caused by the discrepancy between the maximum allowed layer height in the Extruder settings and the layer height at the Print settings. The issue only happens for the MMU2 Draft (0.35mm layer height) profile.

If I set the maximum allowed layer height to 0.35mm at the extruder settings, the project slices correctly with supports.

[image]

palimbro, 2020-03-27:
Yes , it works ;) but, there's an other issue... 
If yout activate automatically supports  or disable just the bridge support  in the picture, the same issue appear again..

Thanks for you quick reply ;)

Provenance : Courrier pour Windows 10

De : Vojtěch Bubník
Envoyé le :vendredi 27 mars 2020 13:01
À : prusa3d/PrusaSlicer
Cc : palimbro; Author
Objet :Re: [prusa3d/PrusaSlicer] MMU2s and supports on prusaslicer 2.2.0issue (#3919)

So the issue is caused by the discrepancy between the maximum allowed layer height in the Extruder settings and the layer height at the Print settings. The issue only happens for the MMU2 Draft (0.35mm layer height) profile.
If I set the maximum allowed layer height to 0.35mm at the extruder settings, the project slices correctly with supports.

lukasmatena, 2020-03-27:
> If yout activate automatically supports  or disable just the bridge support  in the picture, the same issue appear again

Set the maximum strictly higher that 0.35 (e.g. 0.36).
It is a bug and I don't suggest this is the expected workflow. But it should be a workaround for now.

palimbro, 2020-03-27:
That's work ;) thanks again.
Have a great we.
And take care, for you and your family.

Best regards

Sebastien.

lukasmatena, 2020-06-03:
This should be fixed by ed98a859af7c591a7d13aee9d0a0391c258738e2.
Thanks for the report. Closing.

closed 2020-06-03 by lukasmatena