prusa3d/PrusaSlicer #3922: Objects being moved off build plate after slicing

xcschultzx, 2020-03-26:
### Version
2.2

### Operating system type + version
win64

### Behavior
* After slicing, objects are being moved off build plate
* Select *any* User Preset for Printer Settings
* It works properly if a System Preset Printer Setting is used without any changes

[image]

[image]

bubnikv, 2020-03-26:
3MF?

xcschultzx, 2020-03-26:
I just discovered I had a line of starting Gcode moving the Y axis. I've never had an issue during slicing with any Gcode commands, but it is working now.

bubnikv, 2020-03-26:
could you please close this issue then?

closed 2020-03-26 by xcschultzx