prusa3d/PrusaSlicer #3923: Contour perimeter is made over brim

PaulLacatus, 2020-03-26:
### Version
2.2.0

### Operating system type + version
Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3s

### Behavior
The contour perimeter is interfering with brims
just slice a stl and choose brim option

* contour perimeter to be outside brim 
* contour perimeter over brim 
[image]
[t20a.zip](https://github.com/prusa3d/PrusaSlicer/files/4388142/t20a.zip)

It is a bug

#### Project File (.3MF) where problem occurs

Uploaded

neophyl, 2020-03-26:
Its been designed that way.  There are other posts about the exact same thing.

rtyr, 2020-03-26:
Yes, this is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3692, https://github.com/prusa3d/PrusaSlicer/issues/3516, https://github.com/prusa3d/PrusaSlicer/issues/3634 and also related to this feature request https://github.com/prusa3d/PrusaSlicer/issues/724. Closing.

closed 2020-03-26 by rtyr