prusa3d/PrusaSlicer #3925: Cannot add filaments to Ender 3 and custom profile

marouncz, 2020-03-26:
### Version
PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Ender 3 preset and custom printer profiles

### Behavior
I have FRESHLY installed PrusaSlicer V2.2.0.
I went through initial setup wizard and checked the boxes MK3 with 0,4mm nozzle and Ender 3 with 0,4mm nozzle. 
When I select Creality ENDER-3 as current printer, I can only pick from Creality, Generic and Prusament filaments. When I try to add other filaments via  `Filament->Add/Remove Filaments` and  click finish in the filament selection menu, nothing happens, no other filaments were added.
The issue described above can be seen here: https://www.youtube.com/watch?v=ZA-Gzk1oDVA

Same problem occurs when I try to add custom printer profiles exported from same and / or older version of PrusaSlicer. (.ini files from 2.2.0 and 2.1.0 
attached in zip bellow)
[Config bundles.zip](https://github.com/prusa3d/PrusaSlicer/files/4388323/Config.bundles.zip)

neophyl, 2020-03-26:
This is due to the fact that all the other filaments belong to a vendor space that doesnt include those printers.  If you want them you will have to install a compatible printer (such as a mk3) and then make copies of the ones you want remembering to use the detach from vendor button on the filaments dependencies page.
Unless someone knows of an easier way.

marouncz, 2020-03-26:
OK, I can use other filaments after detaching them from system preset.
This works fine, although I find it a little bit annoying. I would expect that after clicking the "Add/Remove filaments" button new filaments would show up. But this is only my personal opinion. :-)
I will probably edit this post to feature request, because I do not know if it counts as a bug.

bubnikv, 2020-03-26:
Please read the section

Changes in handling of system vs. user profiles
of

https://github.com/prusa3d/PrusaSlicer/releases

marouncz, 2020-03-26:
OK, I see.
Thank you very much.

closed 2020-03-26 by marouncz