prusa3d/PrusaSlicer #3926: Feature request: temperature view in slice viewport

liftbag, 2020-03-26:
### Version
_PrusaSlicer version 2.3.0+_
### Operating system type + version
_not relevant_

### Request
It could be useful to have a preview of the printing temperatures when they vary, in the same way as for layers, speed, fan speed etc. For example, to quickly check if mistakes have been made to set up a temp tower.

_This is a new feature request_

rtyr, 2021-02-09:
Related to implementation of https://github.com/prusa3d/PrusaSlicer/issues/3635 and https://github.com/prusa3d/PrusaSlicer/issues/3439. It is now not possible to change the temperature via settings.

rtyr, 2021-09-07:
Implemented in PrusaSlicer 2.4.0-alpha1. Closing.

closed 2021-09-07 by rtyr