prusa3d/PrusaSlicer #3927: Feature request: MMU load/unload toggle for begin and end of print.

Skelten123, 2020-03-26:
Feature request for a new update to Prusa Slicer 2

For the MMU(2s), it would be handy if you don't have to load/unload the filament when using the same color/material with multiple prints. This would help streamline mass printing of nicknacks like for an event or printing the face shield over and over again. This could be a selection in the main window next to the brim toggle for easy access when in advanced and expert modes.

bubnikv, 2020-03-26:
IMHO this should be a firmware request.

Skelten123, 2020-03-26:
Vojtěch Bubník,
I am not sure where the printer code is from about the load/unload. Also, how would I change my request to be a firmware request?

bubnikv, 2020-03-26:
You have to post the issue to https://github.com/prusa3d/prusa-firmware/issues

It may be that a change in PrusaSlicer configuration (not the source code) would be needed once a smarter load / unload is implemented into firmware.

A similar issue that fits this one is 
https://github.com/prusa3d/PrusaSlicer/issues/3807

Ideally we would just emit some special G-codes to the firmware from PrusaSlicer, to which the printer would react based on whether it is a plain or MMU variant.

rtyr, 2021-02-09:
I believe we can close this issue. The discussion can continue in #3807.

closed 2021-02-09 by rtyr