prusa3d/PrusaSlicer #3928: Feature Request: Adding "bridge_flow_ratio" to "Filament Settings - FilamentOverrides"

Whitehawk2000, 2020-03-26:
2.2.0
Win10 Pro 64bit

Because of "bridge_flow_ratio" being filament-dependant, it should be added also to "Filament Overrides" in the Filament Settings Tab.

Thank you very much for the great work!

DanielLaberge, 2021-02-06:
I too wish I could override Bridge Flow and Bridge Speed per filament. 
For instance, Wood PLA requires vastly different settings than normal PLA to bridge properly. 
Right now I'm forced to create copies of my Print Setting profiles just for this filament.

This feature request is in the same spirit, albeit larger in scope: [Arbitrary filament overrides](https://github.com/prusa3d/PrusaSlicer/issues/3067)

GoodyGizmos, 2021-02-12:
Me three!

5jvm0u4, 2024-12-17:
Me four!

5jvm0u4, 2026-09-03:
Is it solved with PS3? My inbox is flooded with this PSBot thing I almost thought it's spam, it's like back in the days when issues would got closed for being stale.