prusa3d/PrusaSlicer #3931: Solid Infill Threshold Area

ybl84f1, 2020-03-27:
### Version
_2.2_

### Operating system type + version
_Windows 10 Pro v1909_

### 3D printer brand / version + firmware version (if known)
_PRUSA MK2, MK3, MINI_

### Behavior
* _Solid Infill Threshold Area should default to 70, but it defaults to 0._
* _Steps needed to reproduce the problem_
  * _View setting in PrusaSlicer_

_Is this a new feature request? No._

rtyr, 2020-03-28:
Can you elaborate? Default PrusaSlicer/Slic3r value actually is 70, but we are using value 0 intentionally in our profiles.

ybl84f1, 2020-03-28:
Okay sorry my bad, when you hover over the box it says the default is 70 but the value is actually zero so I thought that was a mistake and it should be set to 70. I see now that other parameters don't match the "hover" value, I never noticed that before. Apologies.
[image]

rtyr, 2020-03-28:
No problem :-). Closing.

closed 2020-03-28 by rtyr