prusa3d/PrusaSlicer #3933: 3dConnexion Axis Settings not implemented

IronFox, 2020-03-27:
### Version
2.2.0+win64

### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior
Slicer does not heed SpaceMouse speed settings.

Steps: Activate Slider in SpaceMouse settings dialog.
-> Advanced Settings
-> Activate any axis and modify slider/reverse

Expected:
Slicer implements speed/reverse settings immediately or at least upon restart

Actual:
Slicer completely ignores the settings

#### Project File (.3MF) where problem occurs
Anywhere

It's awesome you actually did implement support, but since I also use the space mouse in games, I am used to inverted axis directions. This is no issue in any other application that supports it. You must be doing something differently.

bubnikv, 2020-03-27:
You want all axes to be reversed or just some of them?

Due to the terrible support by 3DConnexion we had to talk to the hardware
directly, so our application is not aware of the settings you do in the
3DConnexon driver. Now after half a year of begging they provided us with
the driver, but now it is too late to change the PrusaSlicer implementation.

IronFox, 2020-03-27:
Oh my. I wasn't aware getting 3d mouse support was this troublesome. Going for the hardware directly does sound like a valid solution although it comes with its own problems, I wager. Have you or will you publish the code that does this? I am still considering adding support in my own programs.

Anyhow, so far I've always inverted all of them. 
To me the space mouse is kind of like moving the camera rather than the object. So if I move the mouse forward I expect the camera to move forward rather then object away from me.

Another, possibly much more useful thing you might want to consider is a possibility to lock the up-axis when rotating via the space mouse.
3ds Max does this, Fusion 360 does not. The difference is quite noticeable.
Looking at an object from a tilted angle is rather uncommon I believe and not having to worry about that angle adds to usability.

bubnikv, 2020-03-27:
> Another, possibly much more useful thing you might want to consider is a possibility to lock the up-axis when rotating via the space mouse.
> 3ds Max does this, Fusion 360 does not. The difference is [...]

Do you have it locked all the time, or just with some modifier key?

Metamere, 2020-03-30:
First off, thanks so much for implementing this.  It's such an important feature for me.  It was confusing at first though, as I didn't expect to have to change the 3D mouse settings in the Prusaslicer application.  I figured it out eventually as nothing I changed in the 3Dxware application affected the operation.  Anyways, yeah, being able to reverse or invert the axes permanently is important.  It's like playing a video game with an analog stick controller where you learned with inverted Y axis, and now you have to learn a completely new control scheme if you can't change it.  Muscle memory can be a pain to have to re-map.  I personally use it with only the zoom axis inverted, but since the other control inputs can be reversed, I'm sure others are used to other axes being inverted.  It sounds like Ironfox inverts all of them.  I could get used to it, but I wouldn't want to have to switch control schemes when I'm going between CAD and the slicer, so I'd have to change my control scheme across the board, which would hurt my productivity in CAD for a while.  Or, I guess I'd probably just learn to be a switch hitter, since my productivity in the slicer isn't as important.
You mentioned the possibility of being able to lock an axis with a keypress.  That would actually be a really nice to have feature, and doesn't seem like it would be too tough to implement now that the basics are in place.  I use that feature in CAD where I can lock all rotation axes so that I can orient the view to a plane, and then pan and zoom to where I need while keeping the view stuck normal to that plane.  Having assignable lock keys for each axis or combinations of axes would be great.  Having the option to hold the key to lock or press and release to toggle the lock would be nice as well.

bubnikv, 2020-10-09:
PrusaSlicer newly communicates with the 3DConnexion devices using the 3DConnexion driver, thus it is possible to configure 3DConnexion for PrusaSlicer in the driver. For the 3DConnexion driver to recognize PrusaSlicer, the user has to copy PrusaSlicer.xml into c:\Program Files\3Dconnexion\3DxWare\3DxWinCore64\Cfg\ . We are actively communicating with 3DConnexion to add our configuration file into their installer 

It will be part of the upcoming PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-09 by bubnikv