prusa3d/PrusaSlicer #3934: Filament settings disappeared on upgrade

Geoffree3, 2020-03-27:
### Version
_Version of PrusaSlicer used goes here_
2.2.0+
### Operating system type + version
MacOS 10.14.6
### 3D printer brand / version + firmware version (if known)
Wanhao i3 Plus
### Behavior
On upgrade to this version, the user filament settings are missing (apart from 1 which is named the same as the printer).  Can I retrieve the previous settings?

rtyr, 2020-03-28:
There are some changes in handling of system vs user profiles since 2.2.0-rc. You can find detailed description in release notes for PrusaSlicer 2.2.0-rc here https://github.com/prusa3d/PrusaSlicer/releases.

Shorter version:
In the PrusaSlicer 2.2.0 series we made a breaking change in handling compatibility between vendor profiles, vendor derived user profiles and user profiles (derived from the "-- default --" profile). Starting with PrusaSlicer 2.2.0, system profiles of one vendor and user profiles derived from a system profile of the same vendor are suppressed if a Print profile of different vendor or a user Print profile (derived from the "-- default --" profile) is selected. To use Print or Filament profiles derived from Prusa3D system profiles together with your user Print profile, you now have to detach these Print or Filament profiles by using  a "detach" button.

When "Show incompatible print and filament presets" is enabled in preferences, PrusaSlicer allows showing incompatible presets at their respective parameter tabs by toggling the "Green flag" icon at the preset tab to a "Red flag". We implemented this feature mainly to help us debugging the profile print and filament compatibility conditions, however some of our users learned to use this feature to work around profile compatibility issues. 

So you can use this parameter to make a "detached" copy of our Prusa3D filament profile which will be visible for your printer.

[image]

marcottt, 2020-03-31:
solved same problem. thanks. But why did attach a material profile to a specific printer ? seem bad

rtyr, 2020-03-31:
See the following discussion https://github.com/prusa3d/PrusaSlicer/issues/3729#issuecomment-592924282. 
Other than that, filament profiles have some printer specific parameters. For example filament overrides (specific retraction settings for particular filament), volumetric speed limits, linear advance, etc.  and even basic settings like temperature, cooling, ... are not 100% universal.

johnbiggs, 2020-04-23:
There was already logic in the profile to read printer notes for compatibility. It was great because new logic could be written. This new hidden magic filter is cumbersome.

closed 2026-09-29 by prusaslicer-bot[bot]