prusa3d/PrusaSlicer #3935: After installing PS2.2.0 - BSOD

merc-slk, 2020-03-27:
### Version
2.2.0 Reverted to 2.1.1

### Operating system type + version
Windows 10 Prof 64 Bit Version 1809  
### 3D printbrand / version + firmware version (if known)
MK3S+MMU2S Firmware 3.8.1 
### Behavior
After installing PS 2.2.0 my system crashed and gave me BSOD. Did reboot. No programs running, No PrusaSlicer running..... BSOD after a few minutes.  Reverted to previous back up of system. Installed PS 2.1.1 Win64 Version. Problem solved.  Something to do with Visual C Package.

bubnikv, 2020-03-27:
I am sorry, but BSOD (blue screen of death) could not possibly be caused by an application. Either you hardware, your operating system installation or your drivers are faulty.

merc-slk, 2020-03-27:
Tried again installing PS2.2.0 - BSOD.  No problem with PS2.1.1. Seems to me this is causing the problem since it's the only install I did. OS is running without any problems. OS and drivers are all up to date. Will try on an other machine to verify.

gege2b, 2020-09-09:
Hi there

I have the same problem

BSOD on Win10 but *only* when I'm using PrusaSlicer (2.2.0 - didn't test older versions)

like the OP, my system is up to date and have no issue at all (even with more ressource-eater softwares)

rtyr, 2021-02-09:
Is this issue still valid? Does it happen with 2.3.0?

gege2b, 2021-02-18:
damn sorry, I forgot this post
this was caused by nVidia driver (my system was indeed up to date... I had to downgrade the driver to get rid of the crash)
never had this issue again since

rtyr, 2021-06-08:
No response. Closing.

closed 2021-06-08 by rtyr