prusa3d/PrusaSlicer #3936: Will Not Resume From A Pause

Jflesch61390, 2020-03-27:
### Version
_PrusaSlicer 2.2.0_

### Operating system type + version
_MacOS Mojave 10.14.6-

### 3D printer brand / version + firmware version (if known)
_Prusa MK3S_

### Behavior
* _I put a pause in to a file to add a magnet and when you go to resume after it pauses, it touches down and then stops. It won't resume from the break_

rtyr, 2020-03-27:
I believe this is a firmware issue. But to be sure, I would like to see your gcode first. Can you share it?

Jflesch61390, 2020-03-27:
[Embed Mag Bit Driver JFv3_0.15mm_PLA_MK3S_2h32m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/4395298/Embed.Mag.Bit.Driver.JFv3_0.15mm_PLA_MK3S_2h32m.gcode.zip)

rtyr, 2020-03-27:
Thank you. The gcode looks OK. So it is most likely some firmware issue. Please open issue here https://github.com/prusa3d/Prusa-Firmware. Closing.

closed 2020-03-27 by rtyr