prusa3d/PrusaSlicer #3937: 3DConnexion Space Mouse is ignored in 2.2.0 - Windows

rosscadguy, 2020-03-27:
### Version
2.2.0 Win64

### Operating system type + version
Windows 10 Home - 64 bit
Version 1903
OS Build 18362.720

### 3D printer brand / version + firmware version (if known)
Prusa MK3S/MMU2S

### Behavior
* _Describe the problem_
3DConnexion Space Mouse is ignored
SpaceMouse Wireless PN: 3DX-600044

* _Steps needed to reproduce the problem_
Launch PrusaSlicer / Manipulate the puck

* _Expected Results_
Scene should rotate/pan/zoom according to puck inputs.

* _Actual Results_
The only inputs from the puck that are reflected on the PrusaSlicer scene are:
• TILT FORWARD (away from user)  ► the scene zooms in.
• TILT BACKWARD (towards user)   ► the scene zooms out.
No other puck input results in any scene manipulation.

_Is this a new feature request?_
No.

I have searched the forums and see no repeated reference to the SpaceMouse not working in version 2.2.0.
I have tried every combination of turning the Space Mouse on and off.  Unplugging and Plugging the USB dongle, plugging the SpaceMouse in directly with a USB cable, restarting PrusaSlicer, Rebooting Windows.  I have uninstalled and reinstalled PrusaSlicer 2.2.0.  To no avail.  Behavior is unchanged.

The Space Mouse works properly in Fusion 360 and in other CAD applications.

Thanks for your help
◄ RB ►

bubnikv, 2020-03-27:
We are accessing the 3D mouse directly. It may be, that we are not
interpreting the data from this particular model correctly. We tested 3
different types, it may be that yours is 4rth. Hard to know.

rosscadguy, 2020-03-27:
You can verify the models that you have tested by looking at the bottom of the SpaceMouse.

 

The only two part numbers I can find on the web for this version of the SpaceMouse Wireless are:  3DX-600044,  and 3DX-700066.  Both units look exactly the same.  So I believe the one you’ve tested is one of these.  Can you verify?   (I have attached two photos of mine.)

 

I have also verified that I am running the current drivers from 3DConnexion:  3DXWare v10.6.5

 

Thanks,

RB

 

pá 27. 3. 2020 v 20:32 odesílatel rosscadguy <notifications@github.com <mailto:notifications@github.com> >

MorroWeb, 2020-03-28:
3DX-600047 does work fine without issues. 
What models were tested?

rosscadguy, 2020-03-28:
I have two instances of 3DX-600044.  One at home and one at the office.  (So two different computers, both running Windows 10.)  Neither one operates correctly on PrusaSlicer v2.2.0.

 

I don’t know what the difference is between 0044 and 0047.  It’s hard to imagine that anything that would work on one wouldn’t work on the other.  Especially just the basic functions of pan, zoom, and rotate.  Should be straight-forward?

 

Thanks for the help.

 

◄ RB ►

bubnikv, 2020-03-28:
I don't know what is going on. PrusaSlicer connects to the hardware
directly, so your version of the driver is only relevant if it interferes
with PrusaSlicer, while it should not. I wonder what would happen if you
deinstall the driver and test PrusaSlicer?

bubnikv, 2020-03-29:
> The only two part numbers I can find on the web for this version of the SpaceMouse Wireless are:  3DX-600044,  and 3DX-700066. 

I have just tested on mine, it is 3DX-600044. I have tested with PrusaSlicer 2.2.0 final release. Wired connection works just fine, wireless as well. Now I am totally confused.

xxxTark77xxx, 2020-03-29:
Hello.  I have the same problem where the Space Mouse does not function in PrusaSlicer.  I have done all updates (Windows 10, PrusaSlicer 2.2 and 3Dconnexion 10.6.5) and rebooted.  My Space Mouse version is 3DX-600044.  Control-M settings which didn’t work either.  Any ideas?

xxxTark77xxx, 2020-03-30:
I had a really interesting discovery and a potential fix.  @bubnikv, hopefully this can help you understand what is happening, I don't know enough about device management.

**Troubleshooting**
The 3Dconnexion receiver for my Space Mouse and CADMouse was in USB (port 1) 
I connected a USB cable into my wireless Space Mouse and into a USB (port 2), rebooted PrusaSlicer, and the wireless Space Mouse began fully working in PrusaSlicer (PS from now on), but plugged in.
I unplugged the Space Mouse from the USB (port 2) and it stopped working.
I moved the USB 3Dconnexion receiver to the port that worked when wired (port 2), rebooted PS, and wireless worked!
I moved the USB receiver back to the original USB (port 1), rebooted, and wireless stopped working.
I moved the USB receiver back to where it worked (port 2), rebooted, and wireless still did not work. 

I have a second unused 3Dconnexion universal USB receiver because I have a Space Mouse and CADMouse.
I removed the old receiver and inserted the new unused receiver into USB port 1.  In the 3Dconnexion app, I had to delete the old pairings and repair both devices to the new receiver.
Both wireless devices worked!
I moved the receiver to port 2.  Space Mouse stopped working in PS.
I moved the receiver back to port 1.  Space Mouse still didn’t work in PS.

I removed the 3Dconnexion receiver from the USB port.
In Device Manager, I deleted all Mice and Other Pointing Devices -> HID-compliant mice
I Plugged in the 3Dconnexion receiver and 2 instances appeared of HID-compliant mice
I paired both devices and Space Mouse worked wirelessly in PS!

**Summary of what I'm seeing**
When the Space Mouse is plugged into either USB port, it works repeatedly.
The Space Mouse will only work wirelessly in PS the first time a 3dconnexion receiver is put into a port.  If the receiver is already a registered device the Space Mouse won’t work wirelessly in PS.

**Fix**
It seems the fix is to remove the receiver from the computer, delete the existing HID-compliant mice in Device Manager, and then re-pair the wireless Space Mouse.  And then, don’t move the receiver =)

Cheers!

bubnikv, 2020-03-30:
@rosscadguy @xxxTark77xxx
So if the wireless mouse is connected by an USB wire, then it works with PrusaSlicer?

xxxTark77xxx, 2020-03-30:
If the wireless Space Mouse is plugged in through USB (like when charging), it always works in PrusaSlicer.

To get the wireless Space Mouse to work repeatedly in PrusaSlicer wirelessly, I needed to delete all HID-compliant mice in Device Manager > Other Pointing Devices. Then I needed to pair the Space Mouse to the 3Dconnexion receiver.

bubnikv, 2020-03-30:
@xxxTark77xxx 

Thanks. I will check Blender implementation. The issue with PrusaSlicer and 3DConnexion driver may be, that PrusaSlicer is open source. I will have to talk to the 3DConnexion company.

Without the 3DConnexion driver, we would likely have to open all the USB subports of a dongle, for which some device is paired and expect data to be received from there.

rosscadguy, 2020-03-30:
The driver software for the 3DConnexion Space Mouse Wireless (and probably all their products) is 3DxWare.  It is what empowers the functions of all the extra “function buttons” on the devices.

 

I uninstalled 3DxWare and found that ALL response to the SpaceMouse in PrusaSlicer went away.  (It used to at least zoom in and out before – but that’s all.)  All the “function buttons” went dead too.  I also confirmed that Fusion360 gets no response from the SpaceMouse with 3DxWare uninstalled.

 

I also have the 3DConnexion CAD Mouse Wireless.  The mouse function operates the same even if 3DxWare is uninstalled.  Though its “function buttons” are dead in this case.

 

I reinstalled 3DxWare, and the limited zoom response reappeared in PrusaSlicer, and all normal response returned to Fusion360.  The “function buttons” on both devices returned to normal.

 

I also discovered that in the START menu, there is of course a 3DConnexion list item.  In its submenu there are two options (which I never noticed before) that offer “Stop 3DxWare”, and “Start 3DxWare”.  I invoked those and they cause the same behavior as if the “3DxWare” driver had been completely uninstalled and reinstalled.  If I picked, “Stop 3DxWare”, then all responsiveness in all apps, and all “function buttons” would go away.  (But again the mouse worked normally.)  If I picked, “Start 3DxWare”, then all previous behaviors would return.

 

So it does not appear that uninstalling or disabling 3DxWare opens up any gateways that allow the SpaceMouse Wireless to operate properly in PrusaSlicer.  I’m glad for any further ideas you may have.

 

Thanks for the ongoing interaction and help.

 

Cheers,

 

◄ RB ►

rosscadguy, 2020-03-30:
Using the ideas from Tark77, I simply tried changing the port of the wireless receiver.  Up to now, I’ve been plugged into one of the ports of a USB_3.0 hub.

For the test, I moved the wireless receiver to one of the ports directly on my laptop (of which there are only two – hence the hub).  PrusaSlicer now reacts to the SpaceMouse Wireless correctly.

 

I moved the wireless receiver back to the hub, and the proper functionality went away.

 

I moved the wireless receiver back to the on-board USB hub and the puck works correctly again. 

 

I could never get any “direct connect” situation to work.  I plugged a USB cable into the puck, and then tried both the hub location and the on-board location and neither case would work.

 

So as long as my wireless receiver Is connected to a USB port directly on the laptop, I’m golden.

 

Cheers,

◄ RB ►

bubnikv, 2020-03-31:
As commented here https://forum.prusaprinters.org/forum/prusaslicer/spacemouse-on-prusaslicer-2-2-0/paged/2/#post-203555

The good news is that we have a Windows specific prototype of 3DConnexion support in PrusaSlicer inspired by Blender, which seems to work with the driver (you can swap axes as you wish), if a configuration file is dropped to 

c:\Program Files\3Dconnexion\3DxWare\3DxWinCore64\Cfg\

 

We are negotiating with a 3DConnexion representative, hopefully a config for PrusaSlicer will be added into the next 3DConnexion driver package. That should conclude the 3DConnexion integration.

rosscadguy, 2020-03-31:
Thanks for the further info.

 

As I mentioned in a previous post, I have two computers.  A Laptop at work, and a desktop at home.

 

I got the laptop working yesterday by making sure the wireless receiver is plugged into an on-board USB slot, not into a USB hub.  3DxWare remained installed.

 

At home on my desktop (a Dell Optiplex 7070) I do not have a hub since there are 5 USB ports on-board.  So the wireless receiver is plugged into the motherboard, and yet,  the SpaceMouse Wireless does not work correctly with PrusaSlicer.

 

Using one of your suggestions, I uninstalled the 3DConnexion driver (3DxWare) and discovered that this was enough to get the puck working correctly with PrusaSlicer.   (This did not work on my laptop.)

 

So it seems that there are varying conditions that allow the puck to work correctly on different machines, and the right combination of elements has to be discovered in each case. 

 

We wish you luck and thank you for your efforts in untangling all this.

 

Cheers,
RB

marcus2tts, 2020-05-07:
I have just tried switching the USB port and got the mouse to work in PS; but it's slow and nothing I do will increase the speed, so it's still not really usable for me in PS as there is nothing more frustrating than waiting for an expected response from an input.

bubnikv, 2020-10-09:
PrusaSlicer newly communicates with the 3DConnexion devices using the 3DConnexion driver, thus it is possible to configure 3DConnexion for PrusaSlicer in the driver. For the 3DConnexion driver to recognize PrusaSlicer, the user has to copy PrusaSlicer.xml into c:\Program Files\3Dconnexion\3DxWare\3DxWinCore64\Cfg\ . We are actively communicating with 3DConnexion to add our configuration file into their installer 

It will be part of the upcoming PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-09 by bubnikv

xxxTark77xxx, 2020-10-09:
Thanks for your continued work on this @bubnikv, it's appreciated.  Where can I find the file PrusaSlicer.xml?