prusa3d/PrusaSlicer #3938: Fusion 360 cannot open after PrusaSlicer have been installed/opened on same PC.

Wiradk, 2020-03-27:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 (build 18363), 64-bit edition
System Info attached.
[PrusaSlicerSystemInformation.txt](https://github.com/prusa3d/PrusaSlicer/files/4395635/PrusaSlicerSystemInformation.txt)

### 3D printer brand / version + firmware version (if known)
Not relevant.

### Behavior
Autodesk Fusion 360 cannot open after I installed and used PrusaSlicer.

I have tried all the normal uninstall and install procedures of both Fusion 360 and PrusaSlicer. Still no positive results.
I have also been in contact with Autodesk Support (level 2), and it is now escalated to Fusion 360 development team.

The only solution that works now (it seems) is to make a clean uninstall as described on Autodesks homepage (with deleting all the files and folder in hidden places), uninstall PrusaSlicer, and then reinstall fusion 360 - without installing PrusaSlicer.

But as soon as I install PrusaSlicer again, work in fusion, use prusaslicer to make the G-code, close both pieces of software, set the PC to sleep -> and wake up the PC and try to open Fusion 360 (without opening PrusaSlicer) , than Fusion 360 cannot open with the system error feedback (attached)
[image]

The feedback from Autodesks automatic solution center states that this error is caused by the fact that the PC do not have the prerequested specs (which it have) (attached).
[image]

So I suspect that PrusaSlicer is hijacking the GPU or RAM and/or Fusion cannot take over after PrusaSlicer.

#### Project File (.3MF) where problem occurs
Not relevant.

bubnikv, 2020-03-27:
> So I suspect that PrusaSlicer is hijacking the GPU or RAM and/or Fusion
cannot take over after PrusaSlicer.

I am sorry, but once PrusaSlicer application closes, no matter what it did,
it is the operating system's responsibility to release CPU and GPU memory.
If that does not happen, then it is most likely a driver issue or a
hardware issue.

I don't know how to help you, especially if it is not PrusaSlicer, but
Fusion 360 that is failing.

Wiradk, 2020-03-27:
Drivers should be up to date.
This only happens after PrusaSlicer have been opened on the PC.
Other graphical softwares (mesh, games ect.) can run together with Fusion
360 on this PC.
So the OS/drivers/hardware are able to release GPU/CPU/RAM in other similar
scenarios.
Only exception is the PrusaSlicer...

So even that it isn't PrusaSlicer that fails, I still suspect that
something goes wrong in relation to PrusaSlicer.

As mentioned it have been escalated to Autodesks Development team.
So I hope any conclusions/questions/investigations from Autodesk will be
supported by the developers in here as well.

So please don't close/ignore this issue before I update with Autodesks
input.

bubnikv, 2020-03-28:
> So please don't close/ignore this issue before I update with Autodesks
input.

Weird things happen in the open world (sic!) of windows, hardware vendors
and their drivers. It is much simpler for Apple, who has a full control of
their whole ecosystem.

We can keep this issue open, but I am afraid we will not be able to help
you. If it was PrusaSlicer crashing, we would may have some chance of
debugging the issue on your computer. If it happened on many similar
computers, we even may buy a similar machine to try to reproduce. But here
we have nothing to reproduce.

skellatore, 2020-04-03:
Weird.. I've got PrusaSlicer And Fusion360 open at the same time right now - Latest 2.2.0 (non alpha or RC Build).

but I don't "Install" PrusaSlicer.
I download the latest release.
Create a new folder in c:\program files\Prusa3D ... In this case.. PrusaSlicer2.2.0
Unzip the downloaded Zip to c:\program files\Prusa3D\PrusaSlicer2.2.0
Create a shortcut to prusa-slicer.exe and put it on my desktop.

rtyr, 2021-04-28:
I believe we can close this now.

closed 2021-04-28 by rtyr