prusa3d/PrusaSlicer #3939: PrusaSlicer allows removable device eject before export is complete.

worldgnat, 2020-03-28:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
* _Describe the problem_
** When exporting a sliced project to a USB key, an eject device button appears next to the "Export" button. This is a fantastic new feature, and I love it! But you can press it before the export is complete, leading PrusaSlicer to fail to export the entire file. Then you have to remove and reinsert the device, re-export, and wait for it to finish. 
* _Expected Results_
** The eject button should be disabled until file export is complete. 
* _Actual Results_
** The eject button is always visible, regardless of export status, meaning that the export process can be interrupted. 

_Is this a new feature request?_
No, I don't think so. From a user experience standpoint, I would argue that this is a bug. 

#### Project File (.3MF) where problem occurs
This is a problem with a feature that does not depend on which model is being sliced. If having these an example project file and screenshots would be helpful, please comment on this issue and I will provide them.

bubnikv, 2020-03-28:
that is a good point.

rtyr, 2021-02-09:
Fixed in 2.3.0. The eject button is now part of the notification. Closing.

closed 2021-02-09 by rtyr