prusa3d/PrusaSlicer #3940: PrusaSlicer defaults to random materials, which is confusing for new users

worldgnat, 2020-03-28:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa Sl1

### Behavior
* _Describe the problem_
** When slicing using a fresh install of PrusaSlicer, the default material is the material which sorts to the top of the list (in my case, 3DM ABS). However, the resin that the printer ships with is Prusa Orange Tough. This resulted in several failed prints before I realized what was going on. 
* _Expected Results_
** Prusa Slicer should default to the material that ships with the printer. For the MK3S, this should be Prusament PLA. For the SL1, this should be Prusa Orange Tough. 
** I realize that Prusa - as of Prusa Slicer version 2.2.0 - is building a general-purpose slicer that supports third party printers (which I fully support!). Presumably, "ships-with material" would be a useful addition to the profiles that are associated with each printer. 
* _Actual Results_
** Prusa Slicer's default material is whatever material sorts to the top of the list alphabetically. 

_Is this a new feature request?_
Yes. 

#### Project File (.3MF) where problem occurs
This is a general feature issue, and doesn't relate to a specific model.

rtyr, 2020-03-28:
Thank you for the report. It is issue in the configuration bundle, not in the PrusaSlicer itself. I will fix it in the next configuration update. All SLA materials were renamed for 2.2.0, this issue is related to it.

rtyr, 2020-04-17:
Fixed.

closed 2020-04-17 by rtyr