prusa3d/PrusaSlicer #3941: Slicing error

AndWik, 2020-03-28:
### Version
Prusa Slicer 2.2.0

### Operating system type + version
Windows 10 Pro 64-bit

### 3D printer brand / version + firmware version (if known)
MK3S but it doesn't matter
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

Before slicing. As it should look:
[image]

After slicing:
[image]

I haven't seen this issue before. Using PrusaSlicer many times/day

Slicing in Simplify3D works perfect.

The code below is OpenScad
#### Project File (.3MF) where problem occurs

$fn=100;

module WallPart () {
	MK = 2;
	minkowski () {
		translate ([0,0,80/2]) cube ([16.5-MK,3.5-MK,80-MK], center=true);
		sphere (d=MK, $fn=50);
	}

	rotate ([90,0,0])
		minkowski () {
			translate ([0,1.49,45/2]) cube ([16.5-MK,4-MK,45-MK], center=true);
			sphere (d=MK, $fn=50);
		}

	translate ([0,-1.5,3])
		rotate ([0,90,0])
			minkowski () {
				rotate ([0,0,45])
					cube ([4-MK,5-MK,16-MK], center=true);
				sphere (d=MK, $fn=50);
			}
}

module OuterPart () {
	MK = 2;
	minkowski () {
		translate ([0,0,130]) cube ([16.5-MK,3.5-MK,160-MK], center=true);
		sphere (d=MK, $fn=50);
	}
}
module Holder () {
	MK = 2
	;
	minkowski () {
		linear_extrude (height =20, center=true) {
			difference () {
				polygon (points=[[-2,0],[-2,45],[190,45],[190,35]]);
		
				scale ([0.6,0.425,1])
					polygon (points=[
					[ 500.00, 0],
					[ 500.00,  0085.88],
					[ 480.00,  0087.36],
					[ 460.00,  0088.64],
					[ 440.00,  0089.72],
					[ 420.00,  0090.57],
					[ 400.00,  0091.17],
					[ 380.00,  0091.52],
					[ 360.00,  0091.63],
					[ 340.00,  0091.51],
					[ 320.00,  0091.19],
					[ 300.00,  0090.68],
					[ 280.00,  0090.00],
					[ 260.00,  0089.08],
					[ 240.00,  0087.83],
					[ 220.00,  0086.14],
					[ 200.00,  0083.92],
					[ 180.00,  0081.07],
					[ 160.00,  0077.57],
					[ 140.00,  0073.44],
					[ 120.00,  0068.62],
					[ 100.00,  0063.00],
					[  80.00,  0056.43],
					[  60.00,  0048.76],
					[  50.00,  0044.28],
					[  40.00,  0039.13],
					[  30.00,  0033.02],
					[  20.00,  0025.37],
					[  12.00,  0017.86],
					[   8.00,  0013.74],
					[   4.00,  0008.92],
					[   2.00,  0005.80],
					[   1.00,  0003.73],
					[   0.50,  0002.34],
					[0.00000,  0.00000]]);
			}
		}
		sphere (d=MK, $fn=50);
	}
}

module ScrewHole () {
	translate ([0,0,0])
		rotate ([90,0,0])
			cylinder (d=4.3, h=50);
	translate ([0,-6,0])
		rotate ([90,0,0])
			cylinder (d=8, h=50);
	translate ([0,0,0])
		rotate ([-90,0,0])
			cylinder (d=8.2, h=10 ,$fn=6);
}

difference () {
	translate ([0,-42,3])
		rotate ([0,-90,0])
			Holder ();
	WallPart ();
	OuterPart ();
	translate ([0,0,70]) ScrewHole ();
	translate ([0,0,130]) ScrewHole ();
	
	translate ([0,-15,0])
			cylinder (d=16, h=6);
	translate ([0,-15,6])
			cylinder (d1=16, d2=0.1, h=5);

	translate ([0,-2,170])
		cube ([4.5,3,70], center=true);
}

//Support
intersection () {
	translate ([0,-30,3]) cube ([12,60,6], center=true);
	union () {
		translate ([-3.5,0,0.47])
			rotate ([90,0,0])
				scale ([1,3,1])
					cylinder (d=2, h=42);
		translate ([3.5,0,0.475])
			rotate ([90,0,0])
				scale ([1,3,1])
					cylinder (d=2, h=42);
	}
}

lukasmatena, 2020-03-28:
Please, provide a 3MF. Thanks.

AndWik, 2020-03-28:
Attached

 

Thanks for your quick response

 

Mr Anders Wikström
Sweden

tonycstech, 2020-03-29:
Looks like model problem. Most people dont understand that slicers dont see models same way human eye does.
I'd like to see the STL file to evaluate and point out the issues i find.
Then developer can address those issues.

FidelCapo, 2021-02-01:
I can't test the fix for this model because there is no stl file available. However, according to tests of other models, this issue should be corrected in next release.
Closing.

bubnikv, 2021-09-22:
no response, closing.

closed 2021-09-22 by bubnikv