prusa3d/PrusaSlicer #3942: Cereal dependency problems in Ubuntu

d01010101, 2020-03-28:
# Version
current shapshot

### Operating system type + version
Linux Mint 19 Tara based on based on Ubuntu 18.04 LTS

### Behavior
Possibly related to https://github.com/prusa3d/PrusaSlicer/issues/3547.

I have installed libcereal-dev, there is the file /usr/include/cereal/cereal.hpp whose presence is verified by the patch in https://github.com/prusa3d/PrusaSlicer/issues/3547 but cmake is stopped anyway with the following message

-- Could NOT find cereal (missing: cereal_DIR)
CMake Error at cmake/modules/Findcereal.cmake:21 (message):
  Cereal library not found.  Please install the dependency.
Call Stack (most recent call first):
  CMakeLists.txt:407 (find_package)

In Findcereal.cmake:

CHECK_INCLUDE_FILE_CXX("cereal/cereal.hpp" HAVE_CEREAL_H)

bubnikv, 2020-03-29:
Do you think you can help find a solution? It would help us a lot. There
are literally hundreds of Linux distros with various quirks and various
versions of libraries. It is not in our powers to solve compilation issues
on all distros, this is usually the responsibility of the distro package
manager.

d01010101, 2020-03-29:
I have finally installed a binary. Feel free to close this issue, as it is no more a problem for me. If it works in other instances of Ubuntu 18, I would just suggest to close.

Otherwise, I can help you if you wish.

bubnikv, 2020-03-29:
It is up to you. If you provide us a solution to your problem, we will
review it and possibly integrate.

Otherwise please close this issue.

closed 2020-03-29 by d01010101

thinkley, 2020-04-02:
Download cereal from github: https://github.com/USCiLab/cereal

Copy the cereal/include/cereal/ directory to Slic3r/cmake/modules/

Error should go away.

/Tom

rickrcomm, 2020-04-02:
same problem. Could you please clarify the source and destination directories? I don't have "Slic3r/cmake/modules/" in my build tree anywhere.

[treefile.txt](https://github.com/prusa3d/PrusaSlicer/files/4424016/treefile.txt)

Thanks

thinkley, 2020-04-06:
Apologies, turns out I was compiling a very old Slic3rPE build.

It may still work if you copy the "cereal/include/cereal/" dir to "Prusa/PrusaSlicer-master/cmake/modules/"

/Tom

mikolaszuza, 2020-07-31:
This worked for me

    Clone cereal, and switch to the develop branch:

git clone https://github.com/USCiLab/cereal.git && cd cereal
git checkout develop

    Build and install the headers:

mkdir -p build && cd build
cmake .. -DJUST_INSTALL_CEREAL=ON
make -j4

    Actually move the headers to the install directory:

sudo make install -j4

guillaume-larche, 2020-08-13:
> This worked for me
>
> Clone cereal, and switch to the develop branch:
>
> ```
> git clone https://github.com/USCiLab/cereal.git && cd cereal
> git checkout develop
> ```
>
> Build and install the headers:
>
> ``` [...]

Works for me too ! Thanks.

bubnikv, 2020-08-13:
If your issue is resolved, could you please close the ticket?

rickrcomm, 2020-08-13:
Will try again to build this week. I will close if successful. 

Rick

metebalci, 2020-09-08:
I think this error is only due to the cache of cmake. If you run cmake, receive the error, and then install libcereal-dev (apt repo), and then run cmake again in the same dir, it gives the same error. If you do cmake in a clean dir after installing the libcereal-dev package, there is no error. Checked this on 18.04.