prusa3d/PrusaSlicer #3943: [Improvement] Wipe tower without sparse layers

jwillmer, 2020-03-28:
The most error prone process in 3D printing is that the first layer is printed perfectly - at least for me. After the first layer is done the print normally succeeds without any issues.  I therefore suggest that the first layer(s) is also printed for the wipe tower regardless of a tool change. Would be really annoying if I have a tool change at the end of the print and the first layer of the wipe tower fails.

I'm not sure if this is already how it works. In that case please close this issue.

bubnikv, 2020-03-29:
It works that way.

With all respect, why didn't you just test before opening a ticket? Please understand, that processing such issues that are not takes energy from the developers, that could be used for fixing real issues and implementing new stuff that you guys are crying for.

closed 2020-03-29 by bubnikv

jwillmer, 2020-03-29:
@bubnikv I know this ticket is annoying if it is already implemented but please understand that testing it on the printer takes up much more time when asking (and answering). Also now the behavior is documented 🥇