prusa3d/PrusaSlicer #3944: Sequential printing disable Pint Mode +

samuelemarazzita, 2020-03-28:
### Version
2.2.0+

### Operating system type + version
Mac OS Catalina 10.15.4

### 3D printer brand / version + firmware version (if known)
Prusa MK3S - FIRMWARE 3.8.1

### Behavior
* When I activate the sequential printing, the plus button to create a pause or a custom G-code doesn't appear. To make it appear, I need to disable the sequential printing, insert the pause and the reactivate it back.
[image]
[image]

bubnikv, 2020-03-29:
that is on purpose. the pause print does not work in sequential print. How
should it work?

samuelemarazzita, 2020-03-29:
Well, if I have 3 pieces and let's say I need to place magnets in one of it, I need to pause the print. The problem is that when I use sequential print, the preview is not following the sequential order (anyway when I place the G-code in Simplyfy3D I see that the route of the extruder works as it has to do).

jonathanwin, 2023-08-30:
+1 for adding support for pause/custom gcode/colorchage in sequential prints (I guess they would need to be assigned to the individual objects and/or instances rather than to the overall file)

jeanlescure, 2024-09-07:
I also need this feature, it would allow me to save 9min between each object print.

To give better context, I've run a stopwatch to time how long it takes between the end of one print and the start of the first layer on the next print. It takes about 11min, of which only 2min are bed leveling and purge line, the rest is pretty much bed/nozzle heating up (I use high temp. filaments).

What I want to do is have 9 copies of my object being printed in sequence, but add a pause and custom gcode whenever it finishes printing an object.

The pause would allow me to just snap the latest print off of the build plate as well as pull off the purge line, then after the pause the custom gcode would run bed leveling and purge line again right before starting on the next object. No waiting for bed/nozzle heating.

Just by being able to do this I would potentially save pretty much 81min of wait time for every 9 of these objects that I print!

I hope this example helps you visualize how this feature can be extremely valuable for us users :)

hn3000, 2026-09-02:
In version 3.0.0-alpha11 I activated the "Complete individual objects" option.

[image]

With that option selected, it is still possible to select "Pause" from the menu on the layer-slider in preview mode, and it no longer gives an error message, but the exported gcode doesn't contain a pause at the selected point (I checked for M601 and the text from the dialog) -- so this doesn't seem to be fixed, yet.

[image]

[image]