prusa3d/PrusaSlicer #3946: Skirt/Priming Area Around Dummy Object or Modifier

TD5023, 2020-03-29:
### Version
2.2.0

### Operating system type + version
Windows 10 Professional

### 3D printer brand / version + firmware version (if known)
Any

### Requested Behavior
I'm wondering if it's possible to add a feature where the user can place a modifier or non-printed object whose sole purpose is for pre-print priming. It would be very useful in situations like this:

[image]
[image]

Here, the objects literally take up the entire print bed and leave next to no room for an outer skirt. Ideally, I would like to place a simple object in the middle of the bed since that's all empty space and use my current skirt settings around it only (while not printing the object itself).

The more I thought about it, the more I realized it would be useful for regular objects, too. When my filament first clears the nozzle, it tends to drag loose strings around the enclosed area of the skirt, which shows on my base layer. If I could move the priming away from the object instead of around it, prints would be cleaner.

I didn't want to tag this as a new feature yet because I didn't know if there was something to handle this that I just couldn't find, but I didn't see anything in the settings or here about it. If I'm just blind, please let me know.

bubnikv, 2020-03-29:
The way it is done at our printer profiles is we have a priming line
scripted in the start G-code. You may most likely use the same trick with
your custom printer. Just install Prusa3D MK3 printer profile and check the
start G-code.

TD5023, 2020-03-29:
That can work, but I was really hoping there was some way to automate the process a little more so that differences would be controlled print-to-print in the plater instead of manually altering g-code. In my above example, a default line that would admittedly work for 90% of prints would need revamped to fit inside of the parts. I've done other prints recently where most of the print bed was covered, so a simple straight-line prime move probably wouldn't fit well anywhere, but a custom shape could be made to fit in an empty spot. 

My goal is to still calculate a priming operation as a skirt (so it can hold a shape while making its minimum extrusion length), but with a little more control over its placement. Overall, though, this is a very minor gripe that is generally pretty easy to work around. If it isn't possible (or worth the time to try and implement), please go ahead and close this and I can make do with g-code changes.

closed 2026-09-29 by prusaslicer-bot[bot]