prusa3d/PrusaSlicer #3947: [FR] Add setting for --datadir in Preferences

GibsDev, 2020-03-29:
I have my print/printer/filament settings stored on my NAS so that I know I'm using the same settings when I'm on different machines in my house. I'm currently using a workaround and specifying --datadir on the command line to overcome this problem.

I need to make sure to launch the program with the --datadir option every time, which gets tedious. More importantly I cant use Fusion360's "Send to 3D Print Utility" feature. The problem is there is no way to inject the --datadir option through Fusion's interface.

bubnikv, 2020-03-29:
You can add a parameter to windows shortcut.

You can create a .bat file and start that from fusion360.

GibsDev, 2020-03-29:
I have a shortcut with the parameter, but Fusion360 does not acknowledge the params added by the shortcut.

I did end up using a .sh file (with git bash installed) similar to your .bat file workaround. The only thing is you can't directly select the .bat or .sh file. You need to navigate to its location and type the filename in the windows open prompt. Thankfully Fusion saves this, so you don't need to do it every time, but it's not an obvious solution.

GibsDev, 2020-03-29:
I guess I should document my script so that other people looking for a solution can use it.

This is the contents of my run.sh script:
`"<path to PrusaSlicer>" --datadir "<path to configurations>" $@ &`

PaulRitter, 2021-05-01:
bumping this. please PLEASE add this. its such a small feature, but it would make the world of difference

bhs67, 2021-12-13:
The PursaSlicer Printer Setting files should be placed in the location of our choosing. This can be accomplished by allowing us to enter a path to all of our unique files.

All of my unique files are on the D: drive (Excel files, Word files, Visual Studio code files, etc).

I periodically backup all of my unique files. If the C: drive crashed, or if the laptop is stolen, it would be easy to retrieve all of my unique files via the backup.

Currently, the Printer Setting files (filename.ini) are saved on C:\Users...\AppData\Roaming\PrusaSlicer\print.

The Printer Setting files should be saved in an area that is easy for us to backup. Allowing us to enter the path to place these files would be very helpful.

Many programs provide a path entry under Preferences, etc. It is a standard menu option. It would be good if PrusaSlicer did the same.

jordanbrown0, 2024-05-15:
+1.  I'd moved my data directory into OneDrive (ref #4443 ) and tweaked the shortcut to point at it, and was happy for a long time, until I installed 2.7.4, it replaced the shortcut, and my settings all "disappeared".  I found them again and tweaked the shortcut again, but a few minutes of panic and annoyance would have been saved if the data directory location was "sticky".  (But, other than moving the data directory to someplace that OneDrive backs up, I don't know how to prevent that panic when I need to restore onto a new computer.)

sochartgit, 2025-01-16:
Up, tried with --datadir and with mklink but I think it's not a big deal to add the setting in PS instead of trying to hack the thing.
Please guys, many of us are asking.

closed 2026-09-29 by prusaslicer-bot[bot]