prusa3d/PrusaSlicer #3949: Prusa Slicer 2.2 now creates toolpaths destroying prints

Seikilos, 2020-03-29:
### Version
New issue since 2.2.0 win 64

### Operating system type + version
Windows 64, 1903

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S 3.8.0

### Behavior
The slicer creates tool paths that seem to peel my prints from the print bed. 

I noticed that having a brim and a skirt creates a tool path that rips the print from the bed.
Elephant Foot compensation might also be a problem (but I doubt that)
I attached a 3mf file where you can see the default behaviour

* Slice the 3mf and look at the toolpath
* The toolpath looks very strange and during print the toolpath gets so bad that the second layer has loose strands that get in the way of the printhead resulting ripping of the print or printing in air
* Disabling skirt when brim is active seems to solve the issue

This behaviour is new to me as I have already used brim without any issues.

#### Project File (.3MF) where problem occurs
[Toolpath elephant.zip](https://github.com/prusa3d/PrusaSlicer/files/4398494/Toolpath.elephant.zip)

bubnikv, 2020-03-29:
Maybe the elephant foot compensation is not a good idea for this kid of object and you should disable it?

@rtyr would you please try to print it? Thank you.

Seikilos, 2020-03-29:
Tried it. Elepephant Foot has no effect. The weird skirt path seems to be the problem.

bubnikv, 2020-03-29:
: The toolpath looks very strange and during print the toolpath gets so bad that the second layer has loose strands that get in the way of the printhead resulting ripping of the print or printing in air

I see, I did  not understand. 

> second layer has loose strands

Second layer of what? The skirt or the brim?

Seikilos, 2020-03-29:
It was hard to see. I think the problem is how the slicer creates the skirt path through the brim: 
[image]

I am very sure that this is a bug because I clearly remember, that the skirt was always created on the outer perimeter of the brim before. 

Now the slicer creates a path where the skirt just plows through the brim. Previously the brim was a continuous path but now the skirt interrupts it. While it does not _directly_ crosses the path, it moves through a location where many extruded lines end (marked in red). This sometimes cause to rip the layer of the bed or create inconsistencies.

I cleaned the bed with soap and printed with PETG, so I do not assume adhesion is the problem there because usually after a soap cleaning I barely can get PETG off (I know, windex, etc)

By looking again at the first layer I am also no longer sure that elephant foot compensation has no effect there as well. It seems that in addition to the skirt+brim chaos the printed object is shrunken and does not touch the brim any more.

I think there are multiple effects in combination that cause the issue.

bubnikv, 2020-03-29:
AFAIK you may just increase the distance of the skirt from the object, so
that the two will no more overlap.

lukasmatena, 2020-03-29:
> I am very sure that this is a bug because I clearly remember, that the skirt was always created on the outer perimeter of the brim before.

It is not a bug. The previous behaviour was in fact inconsistent with settings (see #724). The new behaviour solves two issues for people using skirt as draft shield:
1. they want it close to the object - and that was not possible if the object had brim
2. the tall skirt easily detaches from the bed - possibility to connect it with brim helps.

If you want the skirt further from the object, increase skirt distance in Print Settings.

However, maybe the current solution is not the best idea. Maybe the brim lines should be printed the way they were (i.e., continuously) and the skirt lines should be trimmed. Currently it's the other way round - brim is trimmed by skirt.

Seikilos, 2020-03-29:
> It is not a bug. The previous behaviour was in fact inconsistent with settings (see #724). The new behaviour solves two issues for people using skirt as draft shield:
>
>     1. they want it close to [...]

So it's a feature (albeit a not yet optimal one) and not a bug? :)
This would then imply that the elephant foot itself would be the issue then?

rtyr, 2020-03-29:
Test results looks OK to me. (PLA - based on provided 3MF).

[image]

bubnikv, 2020-03-29:
> So it's a feature (albeit a not yet optimal one) and not a bug? :)
> This would then imply that the elephant foot itself would be the issue then?

There are things that take some time to fine tune. We believe the feature works. Maybe one shoe will fit not all sizes. 

Let's collect some feedback before doing a decision. Maybe you just need to increase squish against the print bed.

This seems to be quite a topic.
https://github.com/prusa3d/PrusaSlicer/issues/3915
https://github.com/prusa3d/PrusaSlicer/issues/3870
https://github.com/prusa3d/PrusaSlicer/issues/3840

Also one customer asks for configurability of the gap
https://github.com/prusa3d/PrusaSlicer/issues/3779

lukasmatena, 2021-08-12:
https://github.com/prusa3d/PrusaSlicer/issues/3870#issuecomment-897501111

closed 2021-08-12 by lukasmatena