prusa3d/PrusaSlicer #3950: Prusaslicer 2.2.0 not responding on win10 pro

mwadeah, 2020-03-29:
### Version
_Version of PrusaSlicer 
(2.2.0)

_Use `About->About PrusaSlicer` for release versions_
cannot open it since the app is not working but it is the latest version downloaded it from the website on 28-Mar-2020

### Operating system type + version
_What OS are you using, and state any version #s_
Win10 pro 64X version 1909
GPU : Zotac running geforce 1080

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s 3.8.0-2684

### Behavior
*once i launch the application, the screen appears then it freezes immediately and when i click on it it shows not responding every time ( pictures will be attached )
i tried un-installing it and installing an older version and same issue 
--the changes done on my pc were nothing except i installed battle.net from blizzard in order to play the new call of duty modern warfare .. which also not working properly XD, that is the only change that i know of before this issue appeared before that the application was running perfectly

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

it just stays like this without responding or sometimes it would show (NOT RESPONDING)

tonycstech, 2020-03-29:
Does the old version work
You may have missing windows components such as visual basic, net framework etc. They come in a number of different versions and year releases. I have them all installed just in case. No problems.
Look into event viewer and see what happens when your load the PrusaSlicer.

Surgikill, 2020-03-30:
I have this same issue. It was working earlier today and now it's not working. I did not uninstall anything.

mwadeah, 2020-03-30:
> Does the old version work
> You may have missing windows components such as visual basic, net framework etc. They come in a number of different versions and year releases. I have them all installed just in case. No problems.
> Look into event viewer and see what happens when your load the PrusaSlicer.

any older version is not working any more 

and this is what goes on in the event viewer
[image]

mwadeah, 2020-03-30:
> I have this same issue. It was working earlier today and now it's not working. I did not uninstall anything.

same with me .. i was actually using it and 30 mins later it stopped working

Surgikill, 2020-03-30:
This is what I get from the security and maintenance control panel. 

> Source
PrusaSlicer

Summary
Stopped responding and was closed

Date
‎3/‎29/‎2020 20:37

Status
Not reported

Description
A problem caused this program to stop interacting with Windows.
Faulting Application Path:	C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe

Problem signature
Problem Event Name:	AppHangB1
Application Name:	prusa-slicer.exe
Application Version:	2.2.0.0
Application Timestamp:	5e75ff9e
Hang Signature:	0fd0
Hang Type:	134217728
OS Version:	10.0.18363.2.0.0.256.48
Locale ID:	1033
Additional Hang Signature 1:	0fd04225cfd6823c89fbccf37224ec06
Additional Hang Signature 2:	7313
Additional Hang Signature 3:	73135cde2901448fd568561764a9486c
Additional Hang Signature 4:	0fd0
Additional Hang Signature 5:	0fd04225cfd6823c89fbccf37224ec06
Additional Hang Signature 6:	7313
Additional Hang Signature 7:	73135cde2901448fd568561764a9486c

Files that help describe the problem
WERInternalMetadata.xml

Surgikill, 2020-03-30:
I found the issue. Thanks to Ahri on Duet Forums. Looks like it's a Nvidia issue. 

[image]

mwadeah, 2020-03-30:
I think it will work because actually what is happening with me is that it is starting in the middle between my two screens ...

It makes sense, i will try it tomorrw and post my results afterwards

mwadeah, 2020-03-30:
> I found the issue. Thanks to Ahri on Duet Forums. Looks like it's a Nvidia issue.
> 
> [image]

Thank you so much .. i have been looking for an answer for three days and it never occurred to me that it would be because of the two monitors thing 

Thank you again 
I will try it and post back

bubnikv, 2020-03-30:
We have a similar hang up reported, which happens if you wake up your computer after plugging it into a docking station. https://github.com/prusa3d/PrusaSlicer/issues/3726

BirkBinnard, 2020-03-30:
I tested 2.2.0 on both monitors of my dual monitor Win10 system and had no problem. But my system has no video card - I use the Intel video contained in the i7 CPU. So I wonder if this issue is unique to nVida video cards, or if it happens with any 3rd part video.

mwadeah, 2020-03-30:
> I found the issue. Thanks to Ahri on Duet Forums. Looks like it's a Nvidia issue.
> 
> [image]

oh my god yes 
it worked like a charm 
this some next level magic 

thank you very much

closed 2020-03-30 by mwadeah

Surgikill, 2020-03-30:
> 
> 
> I tested 2.2.0 on both monitors of my dual monitor Win10 system and had no problem. But my system has no video card - I use the Intel video contained in the i7 CPU. So I wonder if this issue is unique to nVida video cards, or if it happens with any 3rd part video.

It's unique to nVidia because it's a bug in their drivers.

Robiton, 2020-07-01:
An Update, it seems that this issue also applies to the 5X00 AMD video cards.  This weekend I added a 5700XT (Had a 470 before) and now I have this same issue.  Before I was running 2 monitors, now I am running 4.

selectnone, 2020-07-02:
Ahah!  Now I don't have to reinstall PrusaSlicer to get it to work — again!

So for now, we should never leave the app on a non-primary monitor.  

For reference, here's how to get the app working again:
(as text instead of a screenshot)

```
Open file: 
%AppData%\Roaming\PrusaSlicer\PrusaSlicer.ini

Delete the line that starts with: 
window_mainframe = 

Save file.  PrusaSlicer should now open properly.
```

foreachthing, 2020-07-02:
Instead of editing an ini file, you can just press [ALT], [SPACE] and then select Move and use your cursor keys to move the application into view.
To make this work, you have to make sure the application is in front; use [ALT]+[TAB] to select it first.

foreachthing, 2020-07-02:
@Surgikill 
> It's unique to nVidia because it's a bug in their drivers.

I don't think its' a bug in the driver. The application has to make sure it loads where the user left off. I, too, have this issue with all the programs I make. First, count monitors, then confirm resolution, the move it to the new corresponding location. Best if stored as percentage...

selectnone, 2020-07-02:
> Instead of editing an ini file, you can just press [ALT], [SPACE] and then select Move and use your cursor keys to move the application into view.
> To make this work, you have to make sure the application is in front; use [ALT]+[TAB] to select it first.

No, that won't work:
The problem is that the app is frozen and unresponsive — it is _not_ hidden off-screen.

Manually shifting the window across to the main monitor has no effect; the new position is not stored, as the app continues to be frozen.