prusa3d/PrusaSlicer #3953: Defaults to using PVA for first layer of purge block

samohtxotom, 2020-03-30:
### Version
2.2.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S + MMU2S

### Behavior
When using PVA soluble interface it defaults to using PVA for the first layer of the purge block, using PVA as the first layer is less reliable than using PLA. 

_Is this a new feature request?_
Yes, currently the only way to override this that I'm aware of is applying a PLA extruder to "Support material/raft/skirt extruder", however in this case it then builds the entire support structure with the one extruder, wasting a lot of plastic.
If a feature was added to differentiate the "Support material extruder" and "Raft/skirt" extruder" that would fulfill this request. Thank you

bubnikv, 2020-03-30:
Would you please add pictures, screnshots, 3MFs? I cannot follow what you
are trying to say.

samohtxotom, 2020-03-30:
Sorry I will try be clearer.

This is essentially a feature request for an option to select an extruder for the raft, brim and skirt only. 

For example in this 3mf

[prusapreextruderselection_0.2mm_PLA_MK3SMMU2S_19h59m.zip](https://github.com/prusa3d/PrusaSlicer/files/4401245/prusapreextruderselection_0.2mm_PLA_MK3SMMU2S_19h59m.zip)

the first layer is using PVA+, however I want it to use PLA as it adheres better to the bed. 

To achieve that I assign a PLA extruder to "Support material/raft/skirt extruder" as in this 3mf 

[prusapostextruderselection_0.2mm_PLA_MK3SMMU2S_1d1h18m.zip](https://github.com/prusa3d/PrusaSlicer/files/4403175/prusapostextruderselection_0.2mm_PLA_MK3SMMU2S_1d1h18m.zip)

By adding a "Raft, brim and skirt extruder" it will allow better adhesion to the bed for the first layer while maintaining the plastic and time savings.

lukasmatena, 2021-06-03:
This issue should be fixed for 2.4.0 release ( f6de946 ). The first layer of the wipe tower is never printed with soluble filament (as long as there is another option).

I'll close the issue. Let us know if you consider the issue solved after first public 2.4 alpha is released (in several weeks probably). Thanks.

closed 2021-06-03 by lukasmatena