prusa3d/PrusaSlicer #3954: Filaments tab not shown in Configuration Assistant

zapta, 2020-03-30:
### Version
2.2.0+

### Operating system type + version
Mac OSX 10.14.6

### 3D printer brand / version + firmware version (if known)
Custom, Reprap based, Duet3D controller.

### Behavior
If I install PrusaSlicer from scratch and define only my custom printer, the Configuration Assistant doesn't show the Filaments tab. It does show it if I select a Prusa printer such as Mini.

To reproduce
1. Install prusa slicer from scratch.
2. In the Configuration Assistance wizard, define only a custom printer (e.g. reprap based). Do not check on Prusa, Creality or other third party printers.
3. Look for the Filament tab in the Configuration Assistant, it is not shown even though the custom printer is FDM and uses filaments.

_Is this a new feature request?_

No, it's a bug. Assuming that custom printers don't use filaments.

bubnikv, 2020-03-30:
This is a correct behavior, see the release notes of
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-rc
specifically the section Changes in handling of system vs. user profiles

The filaments have parameters specific to print and printer profiles,
therefore they are not available for a custom printer.

zapta, 2020-04-02:
Prusa team, please consider to include also filament profiles and management for printer models that you don't whitelist. Shouldn't be too difficult to have generic profile for PLA, PETG, TPU, etc that are used as baseline. Also, filament vendors such as ColorFab would be more than happy to provide their own profile for non specified printers.

With the current brand/model whitelisting you are excluding vars majority of the printers used by the community .

closed 2026-09-29 by prusaslicer-bot[bot]