prusa3d/PrusaSlicer #3955: Automaticly add Sacrifical layers option

buzkill420, 2020-03-30:
new feature request

Curious if it's possible/ feasable to have Prusaslicer have an option to automatically add sacrificial layers to prints to avoid needing supports.

 Certain features stop bridging from being a clean option and the post processing- though simple - does add up and makes things a bit more complicated ( ie. potential back and forth between slicer and 3d software...i do have to note that the new "load from disc" feature is amazing for that. so thank you)

`[Here's a video that shows a very nice way and a rough way of what i'm talking about](https://www.youtube.com/watch?v=IVtqAn4oDDE)

Shuflie, 2020-03-31:
I've created sacrificial layers in Prusa slicer before for holes that needed support. Just add a cylinder to your part and make it the same thickness as your layer height. Automatic would be nice, but this lets you do everything without jumping back into CAD or even if you don't have access to the CAD files.  See attached zip file for a .3mf as an example.
[duesenverbinder.zip](https://github.com/prusa3d/PrusaSlicer/files/4409723/duesenverbinder.zip)

rtyr, 2021-02-09:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3302. Closing.

closed 2021-02-09 by rtyr