prusa3d/PrusaSlicer #3956: Windows symlinking issue

adras, 2020-03-30:
### Version
PrusaSlicer 2.2.0

### Operating system
Windows 10

### 3D printer brand / version + firmware version (if known)
Does not apply

### Behavior
I was trying to sync my PrusaSlicer settings between multiple PCs. I decided to use OneDrive for that. Therefore the idea was to create a symlink between the Slicer settings directory and a directory in my OneDriver folder. All I then need is to create the symlinks on all PCs, and settings will be synced automatically by OneDrive.

Details:
Slicer Settings directory: c:\Users\andre\AppData\Roaming\PrusaSlicer\
OneDrive directory: c:\Users\andre\OneDrive\3DPrint\PrusaSlicer\

command used for symlink creation: mklink /H /J ..\..\AppData\Roaming\PrusaSlicer\ PrusaSlicer\ 

Issue:
When starting PrusaSlicer I get the following errors:
boost::filesystem::directory_iterator::construct: Das System kann den angegebenen Pfad nicht finden: "C:\Users\andre\AppData\Roaming\Users\andre\OneDrive\3DPrint\PrusaSlicer\boost::filesystem::directory_iterator::construct: Das System kann den angegebenen Pfad nicht finden: "C:\Users\andre\AppData\Roaming\Users\andre\OneDrive\3DPrint\PrusaSlicer\boost::filesystem::directory_iterator::construct: Das System kann den angegebenen Pfad nicht finden: "C:\Users\andre\AppData\Roaming\Users\andre\OneDrive\3DPrint\PrusaSlicer\boost::filesystem::directory_iterator::construct: Das System kann den angegebenen Pfad nicht finden: "C:\Users\andre\AppData\Roaming\Users\andre\OneDrive\3DPrint\PrusaSlicer\boost::filesystem::directory_iterator::construct: Das System kann den angegebenen Pfad nicht finden: "C:\Users\andre\AppData\Roaming\Users\andre\OneDrive\3DPrint\PrusaSlicer\

"Das System kann den angegebenen Pfad nicht finden" is German for "File not found"

bubnikv, 2020-03-30:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/732

closed 2020-03-30 by bubnikv

bubnikv, 2020-09-01:
Fixed with 683af51685ec21d09be4c0ef9ac8228b38bcb56d.

ripxorip, 2022-05-29:
This seems to be broken for the "Physical Printer" files, it forks for the normal printer preset but not for the physical one?

bubnikv, 2022-05-30:
@[ripxorip](https://github.com/ripxorip)
How do we reproduce your issue?

ripxorip, 2022-05-30:
> Collaborator

Replace one of your physical printer .ini with a symlink to it located somewhere else instead.

ripxorip, 2022-05-30:
> @[ripxorip](https://github.com/ripxorip) How do we reproduce your issue?

Replace one of your physical printer .ini with a symlink to it located somewhere else instead.