prusa3d/PrusaSlicer #3957: Install Prusa from the PrusaSlicer-2.2.0+linux-x64-202003211856.tar.bz2

paulhew69, 2020-03-30:
**Please note, I am not a Linux / Ubuntu guru.**
I like using PrusaSlicer as it works well.

### Version 2.2.0
I have downloaded and extracted the files from the PrusaSlicer-2.2.0+linux-x64-202003211856.tar.bz2 archive. In there is 2 folders and a prusa-slicer file.

Going by a google search on how to install an application from a *.tar.bz2 I need to ....
'Make sure you read a file called INSTALL, INSTALL.txt, README, or something similar if one was extracted.'

There are none of these files in the archive. 
Please can someone guide me on what I need to do.

The appimage is good but it does not integrate so stl's etc can be double clicked and open automatically.

I have even tried building my own from the source, again I find the install procedure incomplete.
I guess it is a given you know how to compile code.

bubnikv, 2020-03-30:
AppImage should integrate if you have the appimage daemon installed. I did not try myself though, I do not use Linux as a daily desktop.

We do not provide any other way to integrate PrusaSlicer into your desktop.

bubnikv, 2020-10-16:
Please install one of the AppImage integration daemons. It should work.
Closing.

closed 2020-10-16 by bubnikv