prusa3d/PrusaSlicer #3960: Tool change call missing wipe and retract options for all extruders

HaythamB, 2020-03-31:
### Version
2.2.0-rc+win64

### Operating system type + version
Windows 10 Home Version 1909

### 3D printer brand / version + firmware version (if known)
Custom Jubilee CoreXY Toolchanger, RRF 2.05.1

### Behavior
After enabling Wipe and/or Retract before layer change for multiple extruder printers, these parameters are not applied upon tool changes, resulting in extra material ooze.

* _Steps needed to reproduce the problem_
1. Enable wipe and/or retract before layer change for any extruder in a multi extruder printer.
2. Slice any file, export Gcode
3. Analyze Gcode output, and observe how wipe/layer retract are applied everywhere except before calling for a new tool (eg T0, T1, etc) past the initial loading of the tool at the beginning of the file.

* _Expected Results_
Wipe and retract for layer change options should be equally applied when selecting new tools, picking up the values for retraction speed and length per extruder and should be parsed into Gcode output.

* _Actual Results_
No wipe and/or retract for layer change occur anywhere before a tool change.

Partial solution: simple macro applied per tool change to always apply a wipe and retract, but this does not allow you to enable/disable these features per extruder, and will always attempt to wipe and retract:
{if previous_extruder >= 0 }
; perform a 5% retraction with a 2.5mm X and Y relative move at defined travel speed
G91
G1 F{travel_speed*60}
G1 X2.5 Y2.5 E-{retract_length[current_extruder]*0.05}
; complete remainder of retraction at defined speed
G1 F{retract_speed[current_extruder]*60} E-{retract_length[current_extruder]*0.95}
; return to absolute coordinates and perimeter printing speed
G90
G1 F{perimeter_speed*60}
{endif}

#### Project File (.3MF) where problem occurs
[image]
[image]
[image]
[tool_change_wipe_retract_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/4406144/tool_change_wipe_retract_issue.zip)

lukasmatena, 2020-03-31:
Pre-toolchage retraction is set independently, and you have set it to zero.
[image]

There is still a bug though - if the pre-toolchange retraction length differs between the extruders, the rectraction and unretraction would not match. But I would consider that as a bit different problem.

Does it solve your issue?

HaythamB, 2020-03-31:
That would solve the retraction issue, but what about the wipe? I've mitigated with the macro for now and seems to be fine.

HaythamB, 2020-03-31:
Here's the macro. Your solution does perfectly well at handling the retraction, but leaves out the wipe process.. I've just defaulted to using the macro instead, and leaving the retraction when tool is disabled field to zero for now. Shall I open a separate issue for the wipe handling?
```{if previous_extruder >= 0}{if wipe[current_extruder] or retract_layer_change[current_extruder] }
; perform tool change{if wipe[current_extruder]} wipe{endif}{if retract_layer_change[current_extruder] } retract{endif}
G91
G1{if wipe[current_extruder]} X2.5 Y2.5{endif}{if retract_layer_change[current_extruder]} E-{retract_length[current_extruder]*0.05}{endif} F{travel_speed*60}
{if retract_layer_change[current_extruder]}; complete remainder of retraction at defined speed
G1 F{retract_speed[current_extruder]*60} E-{retract_length[current_extruder]*0.95}{endif}
{if wipe[current_extruder] or retract_layer_change[current_extruder] }; return to absolute coordinates and perimeter printing speed
G90
G1 F{perimeter_speed*60}{endif}
{endif}{endif}
```

lukasmatena, 2020-04-02:
@HaythamB To be honest, I don't understand the issue with the wipe. When I load your 3MF and add the toolchange retraction (I set it to 5mm), the gcode around the toolchange looks like this:
```
G1 X152.998 Y128.238 E0.12066 ; perimeter
G1 X154.323 Y128.393 E0.09958 ; perimeter
G1 X154.120 Y128.672 F13200.000 ; move inwards before travel
; stop printing object Moai_Body.stl id:0 copy 0
G1 F10560.000
G1 X154.251 Y128.767 E-0.08225 ; wipe and retract
G1 X153.793 Y128.702 E-0.09993 ; wipe and retract
G1 X152.556 Y128.596 E-0.26803 ; wipe and retract
G1 X151.394 Y128.542 E-0.25113 ; wipe and retract
G1 X150.024 Y128.521 E-0.29582 ; wipe and retract
G1 X148.679 Y128.540 E-0.29031 ; wipe and retract
G1 X147.477 Y128.593 E-0.25987 ; wipe and retract
G1 X146.234 Y128.699 E-0.26929 ; wipe and retract
G1 X145.748 Y128.767 E-0.10596 ; wipe and retract
G1 X145.589 Y128.411 E-0.08417 ; wipe and retract
G1 X146.311 Y128.305 E-0.15744 ; wipe and retract
G1 X147.485 Y128.204 E-0.25447 ; wipe and retract
G1 X148.674 Y128.151 E-0.25689 ; wipe and retract
G1 X150.106 Y128.134 E-0.30918 ; wipe and retract
G1 X151.385 Y128.153 E-0.27617 ; wipe and retract
G1 X152.998 Y128.238 E-0.34890 ; wipe and retract
G1 X154.323 Y128.393 E-0.28795 ; wipe and retract
G1 E-1.10224 F2400.00000 ; retract for toolchange
; Filament-specific end gcode 
;END gcode for filament
T0 ; change extruder
; Filament gcode
; printing object Moai_Body.stl id:0 copy 0
G1 X154.996 Y128.668 F13200.000 ; move to first perimeter point
G1 E5.00000 F2400.00000 ; unretract
G1 F3000.000
G1 X155.063 Y128.519 E0.01219 ; perimeter
G1 X155.720 Y128.656 E0.05010 ; perimeter
G1 X156.447 Y128.842 E0.05605 ; perimeter
```
So the extruder is in fact wiped while the filament is being retracted. Sum of all the negative E-moves is 5mm, exactly the value from config.

Your macro (apart from having incorrectly paired ifs and endifs) seems to always move the nozzle diagonally a little bit. You can of course do that if you want to, but I don't think there is something that PrusaSlicer does wrong. Could you please clarify that?

HaythamB, 2020-04-02:
Yes you're correct, I didn't have it set up properly. I tried your suggestion and its working fine. Thanks for the help!

closed 2020-04-02 by HaythamB