prusa3d/PrusaSlicer #3961: Enable nested IF statements within macros please!

HaythamB, 2020-03-31:
### Version
2.2.0-rc+win-64

### Operating system type + version
Windows 10 - 1909

### 3D printer brand / version + firmware version (if known)
Custom Jubilee CoreXY Toolchanger - RRF 2.0.5

### Behavior
Cannot create nest IF THEN ELSE statements within macros. Macro parsing experiencing odd behavior if attempting multi-line IF statements nested within each other (ie no branching can occur within a single IF..THEN segment - can't add another IF statement for any other purpose)

* _Steps needed to reproduce the problem_
1. Create any macro with an IF statement, some code, and another IF statement starting and terminating before the "parent" {endif} code block.
2. GCode file will be generated, but output behavior is erratic. For instance, referring to travel_speed variable is no longer picked up by macro if the line of Gcode commands immediately before the macro output (not comments) has used the travel_speed variable as well. 

* _Expected Results_
Nested IF statements that work like any other scripting language, where parent-child IF statements can exist and mutually access the same variables from PE slicer.

_Is this a new feature request?_
YES! Please! :)

bubnikv, 2020-03-31:
{if ...} 
  {if ...}
  {else}
  {endif}
{else}
{endif}

HaythamB, 2020-03-31:
Yes, my bad! I had incorrectly assumed the output was incorrect.. Thanks!

closed 2020-03-31 by HaythamB