prusa3d/PrusaSlicer #3963: Crash slicing specific model

G-Pereira, 2020-03-31:
[image]

### Version
2.1.1
### Operating system type + version
Fedora 31 - KDE Plasma (updated)

### 3D printer brand / version + firmware version (if known)
Hypercube (not relevant)

### Behavior
Prusa Slicer crashes when slicing a model.
Steps to reproduce:
- Open file (.stl or .amf)
- Click "Slice Now"

### Files
[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/4408888/bug.zip)

rtyr, 2020-03-31:
I am unable to reproduce it on 2.1.1/2.2.0 win64.

G-Pereira, 2020-03-31:
I have tried again an I wasn't able to reproduce also. I find that strange because it crashed ~5 times in a row with that model.
Meanwhile I have power cycled my computer and maybe that changed something.

G-Pereira, 2020-03-31:
Feel free to close this issue if you don't have sufficient data for debug

closed 2020-03-31 by bubnikv