prusa3d/PrusaSlicer #3964: Crash when closing on Linux

wavexx, 2020-03-31:
### Version
Current master (095062e371115c864f7884da2ee927f43ee723f3)

### Operating system type + version
Linux (Debian unstable)

### Behavior
* Open PE
* Add a box
* Slice
* Attempt to quit
* Crash with assertion failure:

``../src/unix/glx11.cpp(584): assert ""xid"" failed in SetCurrent(): window must be shown``

This occurs due to a call to reset_volumes() inside the GLCanvas3D destructor, which is now attempting to activate the context without a valid window.

```
#0  0x00007ffff65eb434 in wxGLContext::SetCurrent(wxGLCanvas const&) const ()
    at /usr/local/lib/libwx_gtk2u_gl-3.1.so.4
#1  0x000055555637bcaa in Slic3r::GUI::GLCanvas3D::_set_current() (this=0x555558501a20)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/GLCanvas3D.cpp:5097
#2  0x0000555556369732 in Slic3r::GUI::GLCanvas3D::reset_volumes() (this=0x555558501a20)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/GLCanvas3D.cpp:1714
#3  0x0000555556368c47 in Slic3r::GUI::GLCanvas3D::~GLCanvas3D()
    (this=0x555558501a20, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/GLCanvas3D.cpp:1605
#4  0x000055555648262f in Slic3r::GUI::View3D::~View3D()
    (this=0x5555584f40f0, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/GUI_Preview.cpp:50
#5  0x0000555556482698 in Slic3r::GUI::View3D::~View3D()
    (this=0x5555584f40f0, __in_chrg=<optimized out>)
...
```

@enricoturri1966

enricoturri1966, 2020-03-31:
@wavexx 

Could you please check if setting:

// Moves GLCanvas3DManager from being a static member of _3DScene to be a normal member of GUI_App
#define ENABLE_NON_STATIC_CANVAS_MANAGER (0 && ENABLE_2_3_0_ALPHA1)

removes the crash ?

wavexx, 2020-03-31:
On Tue, Mar 31 2020, enricoturri1966 wrote:
> @wavexx
>
> Could you please check if setting:
>
> // Moves GLCanvas3DManager from being a static member of _3DScene to be a normal member of GUI_App
> #define ENABLE_NON_STATIC_CANVAS_MANAGER (0 && ENABLE_2_3_0_ALPHA1)

Doesn't build just by changing this in Technologies.hpp
(something else I missed?)

enricoturri1966, 2020-03-31:
@wavexx

Sorry for that, my fault.
e886475fe7b55500d0b5db9e4b1813ab412f4b3b should fix the build when ENABLE_NON_STATIC_CANVAS_MANAGER is disabled.

wavexx, 2020-03-31:
On Tue, Mar 31 2020, enricoturri1966 wrote:
> Sorry for that, my fault.
> e886475 should fix the build when ENABLE_NON_STATIC_CANVAS_MANAGER is
> disabled.

Ok, by disabling ENABLE_NON_STATIC_CANVAS_MANAGER the assertion failure
during quit is gone.

enricoturri1966, 2020-04-01:
@wavexx
Could you please check if 1f3046f65d4f14ad56f571b982d15d1e271128a5 helps in removing the crash when ENABLE_NON_STATIC_CANVAS_MANAGER is enabled ?

wavexx, 2020-04-01:
On Wed, Apr 01 2020, enricoturri1966 wrote:
> @wavexx
> Could you please check if 1f3046f helps in removing the crash when
> ENABLE_NON_STATIC_CANVAS_MANAGER is enabled ?

I rebuilt from a fresh master 1f3046f65d4f14ad56f571b982d15d1e271128a5.
It works correctly.

I'll report a few other findings though. On the same commit (any recent
one really), slicing in a debug build fails with an assertion failure:

prusa-slicer: /home/ydelia/projects/PrusaSlicer/src/libslic3r/GCode/ToolOrdering.cpp:403: void Slic3r::ToolOrdering::fill_wipe_tower_partitions(const Slic3r::PrintConfig&, coordf_t, coordf_t): Assertion `! m_layer_tools.empty() && m_layer_tools.front().has_wipe_tower' failed.

I had to remove the assertion to continue.

Rebuilding on master 8027fc2fb2b242bd6082df12e34e1e1406aadd93 crashes
earlier, just after the "Loading ..." dialog is closed, due to recent
Mouse3DController changes. I think it's relevant, so I'll show the stack
of that too:

#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff62b955b in __GI_abort () at abort.c:79
#2  0x00007ffff7e4880c in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7e538f6 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7e53961 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x0000555555bd5911 in std::thread::~thread() (this=0x5555580d8be0, __in_chrg=<optimized out>)
    at /usr/include/c++/9/thread:139
#6  0x000055555609f9cc in Slic3r::GUI::Mouse3DController::~Mouse3DController()
    (this=0x5555580d8a60, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/Mouse3DController.hpp:28
#7  0x0000555556067f4b in Slic3r::GUI::Plater::priv::~priv() (this=0x5555580d5a80, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/Plater.cpp:2226
#8  0x00005555560ab5d6 in std::default_delete<Slic3r::GUI::Plater::priv>::operator()(Slic3r::GUI::Plater::priv*) const
    (this=0x55555818cc30, __ptr=0x5555580d5a80) at /usr/include/c++/9/bits/unique_ptr.h:81
#9  0x00005555560a5a18 in std::unique_ptr<Slic3r::GUI::Plater::priv, std::default_delete<Slic3r::GUI::Plater::priv> >::~unique_ptr() (this=0x55555818cc30, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/unique_ptr.h:292
#10 0x000055555607942e in Slic3r::GUI::Plater::~Plater() (this=0x55555818c990, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/Plater.cpp:4572
#11 0x0000555556079456 in Slic3r::GUI::Plater::~Plater() (this=0x55555818c990, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/Plater.cpp:4574
#12 0x00007ffff6b5601b in wxBookCtrlBase::DeletePage(unsigned long) () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#13 0x00007ffff6afcaf2 in wxNotebook::DeleteAllPages() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#14 0x00007ffff6afce63 in wxNotebook::~wxNotebook() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#15 0x00007ffff6afcee9 in wxNotebook::~wxNotebook() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#16 0x00007ffff6cc80c8 in wxWindowBase::Destroy() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#17 0x00007ffff6cc80fc in wxWindowBase::DestroyChildren() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#18 0x00007ffff6a722fd in wxWindow::~wxWindow() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#19 0x000055555605340c in wxFrame::~wxFrame() (this=0x555557da7f60, __in_chrg=<optimized out>)
    at /usr/local/include/wx-3.1/wx/gtk/frame.h:16
#20 0x0000555556053eb8 in Slic3r::GUI::DPIAware<wxFrame>::~DPIAware() (this=0x555557da7f60, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/GUI_Utils.hpp:129
#21 0x0000555556056a6a in Slic3r::GUI::MainFrame::~MainFrame() (this=0x555557da7f60, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/MainFrame.hpp:101
#22 0x0000555556056a86 in Slic3r::GUI::MainFrame::~MainFrame() (this=0x555557da7f60, __in_chrg=<optimized out>)
    at /home/ydelia/projects/PrusaSlicer/src/slic3r/GUI/MainFrame.hpp:101
#23 0x00007ffff6b4e826 in wxAppBase::DeleteAllTLWs() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#24 0x00007ffff6b4e8b4 in wxAppBase::CleanUp() () at /usr/local/lib/libwx_gtk2u_core-3.1.so.4
#25 0x00007ffff712b8de in wxEntryCleanup() () at /usr/local/lib/libwx_baseu-3.1.so.4
#26 0x00007ffff712b9fb in wxUninitialize() () at /usr/local/lib/libwx_baseu-3.1.so.4
#27 0x00007ffff712c7ba in wxEntry(int&, wchar_t**) () at /usr/local/lib/libwx_baseu-3.1.so.4
#28 0x000055555568c435 in Slic3r::CLI::run(int, char**) (this=0x7fffffffdd60, argc=3, argv=0x7fffffffdf68)
    at /home/ydelia/projects/PrusaSlicer/src/PrusaSlicer.cpp:533
#29 0x000055555568df44 in main(int, char**) (argc=3, argv=0x7fffffffdf68)
    at /home/ydelia/projects/PrusaSlicer/src/PrusaSlicer.cpp:732

enricoturri1966, 2020-04-01:
@wavexx
Could you please create two new separated issues for your findings so that they get processed and close this one if you consider it solved ?

wavexx, 2020-04-01:
Issue #3971 for the assertion failure.
Rebuilding on the current head made the second go away \o/
Closing this!

closed 2020-04-01 by wavexx