prusa3d/PrusaSlicer #3965: Wipe tower - wrong code to disable linear advance for reprap fw.

marekdovjak, 2020-03-31:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Duet Wifi, reprap fw

### Behavior
When starting wipe tower gcode, linear advance (pressure advance) is disabled. For Marlin fw this is done properly with M900 K0. However, for reprap, slicer ouptuts M572 D0 S0. This is valid only for tool (extruder) 0. When wiping extruder 1, command must be M572 D1 S0 and so on.
problem is on this line https://github.com/prusa3d/PrusaSlicer/blob/a6a77361ba27bd1119548ac06a48b988b98da4c8/src/libslic3r/GCode/WipeTower.cpp#L105

documentation - https://duet3d.dozuki.com/Wiki/Gcode#Section_M572_Set_or_report_extruder_pressure_advance

_Is this a new feature request?_
no

lukasmatena, 2020-04-03:
Thanks. Could you please test that the behaviour is correct in the following build and that the printer really understands the gcode and prints?
https://drive.google.com/open?id=1P5SfMEpkB1pGAu_5NzFFozD9Q-msmDiW

marekdovjak, 2020-04-20:
@lukasmatena I have requested access to your google drive link, still no response.

lukasmatena, 2020-04-20:
@marekdovjak Sorry about that. You should have access now.

marekdovjak, 2020-04-21:
gcode is good now. thanks

lukasmatena, 2020-05-21:
@marekdovjak Thanks for the testing. I just pushed the fix into `master`.
Closing.

closed 2020-05-21 by lukasmatena