prusa3d/PrusaSlicer #3968: Avoid crossing perimeters is broken

tecnopolis-ve, 2020-04-01:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
CR10S with SKR1.3 core

### Behavior
* I'm trying to print a mug with petg, I've checked "avoid crossing perimeters" in order to avoid extruder travel inside the inner cylinder and thus leaving unwanted strings/blobs, this feature is not working as expected. I've figured out how to alleviate this problem by printing infill first and then perimeter.

* _Expected Results_
The extruder movements should be inside walls and never should cross any perimeter (inner or outher)

* _Actual Results_
[image]

_Is this a new feature request?_
This feature already exists but it is broken.

rtyr, 2020-04-01:
There are plenty of similar issues related to this feature. See https://github.com/prusa3d/PrusaSlicer/labels/path%20planning

Photogad, 2020-04-15:
Its "avoid crossing perimeters" not "cant cross perimeters".  As you will see in the gcode preview, with this setting enabled, it will cross the perimeters, far less but still do it in some places. It still helps petg prints a lot

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

No model was provided, so we just hope that our reworked algorithm works better with this model.

tecnopolis-ve, 2020-12-13:
Hello @bubnikv 

PrusaSlicer 2.3.0 Beta3 (released yesterday, december 12)

First of all, I really appreciate the great work, PrusaSlicer is a great tool and I use it every day. I'm here again reporting that the same model with same settings (avoid crossing perimeter: on) is not working as expected.

The main problem is the supports, the travel between support is done across any perimeter, picture:

[image]

Disabling support generation seems to "fix" the problem

[image]

Find attached the model!

[tiki_beer_mug.zip](https://github.com/prusa3d/PrusaSlicer/files/5685264/tiki_beer_mug.zip)

hejllukas, 2020-12-14:
Thank you for reporting. In this case, at least the inner hole's perimeter shouldn't be crossed, so definitely, it's a bug. I'll look at it.

hejllukas, 2020-12-15:
It should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/0ca337ac7a91c183aa8d1fa6d2c288632f4248cb, and will be included in the next release. 
Travel between supports can lead through external perimeters but shouldn´t lead through the perimeter of holes. The result could be seen in the attached screenshot.
[image]

tecnopolis-ve, 2020-12-15:
nice! I'll give a try!

rtyr, 2021-02-09:
@tecnopolis-ve Is it ok? Can we close the issue?

tecnopolis-ve, 2021-02-09:
@rtyr yes! I can confirm that this bug was fixed in the last release

Cheers!

closed 2021-02-09 by tecnopolis-ve