prusa3d/PrusaSlicer #3969: crash if $USER is not set

kopchik, 2020-04-01:
### Version
2.2.0+linux-x64

### Operating system type + version

Ubuntu 18.04 in docker

### Behavior

New prusa slicer crashes if $USER is not set:

```
$ ./PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_S_construct null not valid
  Aborted (core dumped)
```

I believe this is happening in this line: https://github.com/prusa3d/PrusaSlicer/blob/58192ba6c224298bfa9804e016831e77051c1be7/src/slic3r/GUI/RemovableDriveManager.cpp#L231

I think I need to explain why USER wasn't set in my environment. I run Prusa Slicer from docker, and this environment variable simply wasn't set because there was no login process.

clintlombard, 2021-09-21:
This has taken me a few hours to find this... Thank you kind sir

lukasmatena, 2021-09-22:
Should be fixed with d57d3f9 for 2.4.0-alpha2. Please, retest after it is released.
Closing.

closed 2021-09-22 by lukasmatena