prusa3d/PrusaSlicer #3970: Incorrect coordinates on Ender 3

MikiGrit, 2020-04-01:
### Version
2.2.0

### Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3, stock version

### Behavior
When printing any object on my Ender 3 (that has been sliced on PrusaSlicer 2.2.0), the resulting print is not placed as displayed in the program. I've roughly measured, that at the bottom, when program is showing about 2,5cm space to the edge (allowed print area), the physical edge is in fact only about 2cm away. It is also off on other sides, when right side showed about 5cm, in reality it was 6cm to the physical edge...

JohnnyDeer, 2020-04-02:
Hi, can you please provide any 3mf file, we would like to check settings, thank you.

MikiGrit, 2020-04-02:
Hi, here is the 3mf file:
[test_rect.zip](https://github.com/prusa3d/PrusaSlicer/files/4422693/test_rect.zip)

and here is photo of the resulting print:
[image]

It may be a similar issue as Utimaker Cura had: https://github.com/Ultimaker/Cura/issues/6267

espr14, 2020-04-03:
This is based on your [0,0,0] position. This position is, typically, outside of the bed but PrusaSlicer does not take it into account.

Solutions:
1. Move your x/y end stops so the [0,0,0] is on the corner of the bed. This brings hitting bed by nozzle. I recommend adding "G1 Z5" just after G2 (in "Start G-code").
2. Leave end stops so the [0,0,0] is out of the bed. Add to your "Start G-code" after G28:
```
G1 Z5
G1 X5 Y5
G92 X0 Y0 E0
```
According to your measuring you are 5 milimeters off. Otherwise, add more or less to X/Y.

MikiGrit, 2020-04-03:
Ok, thanks for insight and solution 👍. I'll try that.

beikeland, 2020-09-25:
Had similar issue but much more pronounced (`X5 Y15`) with Ender 3 Pro, which triggers warnings in the slicer - is that possible to circumvent or is the best option to start moving the endstops?
[image]

espr14, 2020-09-26:
You can change origin in the PS to -5, -15. This should work without moving the endstops.
[image]

rtyr, 2021-02-09:
I believe this can be closed.

closed 2021-02-09 by rtyr