prusa3d/PrusaSlicer #3972: "Split to Objects" option to retain original position

keeshka, 2020-04-02:
### Version
v2.2.0+

### Operating system type + version
Mac OS X Sierra v10.12.6

### 3D printer brand / version + firmware version (if known)
Stock Original Prusa i3 MK3S

### Behavior
* _Describe the problem_
I’m trying to print a complex architectural model of a building.  The design is imported from the CAD program 'Chief Architect'.  Prior to printing, there are some internal parts that need to be deleted.  The “Split to Objects” feature is perfect to do this.  The individual part is correctly converted into a separate object and then can easily be deleted from the model.  This is exactly what is required.  However, creating individual objects from the model results in ALL newly converted objects being placed on the print bed.  This then ruins the model and it can no longer be printed.

* _Steps needed to reproduce the problem_
Not a problem with PrusaSlicer, "Split to Objects" is behaving as designed.

* _Expected Results_
A feature request to have an option for "Split to Objects" to retain the original position for newly created objects.

* _Actual Results_
n/a

_Is this a new feature request?_
YES

neophyl, 2020-04-02:
Can’t you use split to parts instead ?  I thought that did exactly what you require.

keeshka, 2020-04-02:
I tried.  "Split to Parts" results in one object with many parts.  You cannot delete a part, only an object.  If I try to delete the part, the entire object model is deleted.

neophyl, 2020-04-02:
You should be able to select the sub part by selecting it from the listing on the right and then deleting it. Once selected you can then move, rotate, scale etc as well as delete the individual part.

keeshka, 2020-04-02:
Yes that is true.  In my case, a sliced "floor" of the building design after "Split to Parts" results in over 2000+ parts.  Since you cannot select the individual part on the print bed 3D model (i.e., the entire model can only be selected), you have to scroll thru the sub part listing on the right until the part is highlighted on the print bed, then delete that part from the listing.  Incredibly tedious and time consuming to do with over 2000 parts.  This would SO much easier with "Split to Objects".  Simply select the object on the print bed then delete.  If only the objects would retain their original position so that the print job wouldn't be ruined.  Hence the feature request.  Deleting from the parts list on the right is not obvious and doesn't show up when right-clicking a selected part.  So thank you for pointing this out.

neophyl, 2020-04-02:
Split to part and split to object would still generate the same number of items.  The only difference between the 2 is their logical grouping.  Objects are treated as separate items which is why ALL the separated items drop to the bed as the rule is all objects must at least touch the bed to be printable.

An object made up of different parts only has to have one part touching the bed to be viable (even if that would make some of the parts unprintable.  

If you import a model that has 2000 separate parts then splitting to object even if they didn't drop down would mean you still had 2000 objects that you have to select between so I don't see the functional difference.

Slicer has no way of knowing what your 'objects' would be, its all or nothing.  Have you tried editing your original model and exporting sub sections of it separately ?

You could also post one of your .3mf files here to seek advice (save project as).  At least that way we would have a greater understanding of just what you are trying to achieve.

keeshka, 2020-04-02:
Hi neophyl, and thank you for your explanations.  I've attached a screen grab to illustrate why I've put in this feature request.  The screenshot is an example of a 1st floor house slice.  The parts that need deletion are the soffits from the roof that are lower than the height of the 1st floor (hence they are included in the slice).  And no I cannot lower the slice to not include them as it would lower the 1st floor height and the model proportions would no longer be correct.

In the screen grab, I've split the model into parts.  It then took a while to scroll thru the right hand listing until I found a correct part to highlight (part # 921 is highlighted and would now be deleted).  Repeat until all the unwanted soffit parts are deleted (a very tedious and ugly process).

This would be so much easier if I could just spin the 3D model print bed around, mouse select the part in question, then delete.  "Split to Objects" would be a perfect way to accomplish this if only the parts would stay in their original location.

That is in essence the functional difference.  Manually find (for example) 40 parts scattered throughout a listing of 1000's or point-click-highlight-delete directly on a 3D graphical model.

PS:  Editing of the original design is not an option as these "master" plans are delivered directly from a architectural design company.

[image]

neophyl, 2020-04-02:
You didnt include the .3mf file. A screen grab is of only limited use. 

If you split that to objects then how many objects are created and fall to the bed ?
What format/file type is the 3d model that you imported ?

I'm not a dev on here but personally I think you are asking too much of software whose primary function is to slice a model to generate gcode.  You need a cad package that will let you edit the model prior to importing it.  To easily remove the parts you aren't interested in.  Ergo the enquiry about the file format.  Depending on the format of the model one of the free ones may be all that's required but without further information I cant make any estimation.  

Interesting problem though.

keeshka, 2020-04-03:
Hi neophyl, and thanks again for your help.

re:  "... You need a cad package that will let you edit the model prior to importing it. To easily remove the parts you aren't interested in. ..."

It's not that I disagree with you but it is a bit like a Catch-22 situation.  I don't know what I'm not interested in (e.g., needs to be editing/deleted) until AFTER I cut the imported design in PrusaSlicer.  The screen grab above is one example of what needs to be deleted after the cut.  Impossible to know this ahead of time as I'm not involved with the original design.  Only after I get the design and cut can I see what needs to be deleted.

Without knowing the application technicalities, it seems like a simple concept to (1) split to objects and optionally leave in original position, (2) delete whatever, (3) if necessary, combine back into single object on print bed, then (4) slice, g-code, and print.

Hopeful that Prusa will see this and say, "yup easy to do, will just add a checkbox option".  If not, well, at least I asked.  And I'll try to work around my issue as best I can.

Thanks for responding and offering your insight. I did learn some new things.

keeshka, 2020-04-03:
Whoops, forgot to include the entire model prior to cutting for the reference.  Hope this helps give you the "total" picture and why I need to cut the model prior to printing.

[image]

neophyl, 2020-04-03:
Thanks for that Keeshka.  I notice from your last full on pic that is a obj file so fairly common and most cad software will open those.  Even something like Blender will (those its not the best at obj).  

If I were to do what you want (as I'm moderately familiar with Blender) I'd import it, go into edit mode, turn on wireframe so I selecting parts behind others is possible,, select side elevation view and box select the top and either delete or hide the parts I wanted to remove, preferably with hide.  I could then rotate around and hide or delete the other parts I didn't want, just like you want to do in Prusa Slicer and once I have just what I wanted I would then just select the visible parts and could then export with the 'only selected' option.  The underlying model would all still be there  so I could edit it again later for example if I wanted to just do the upper, but I'd then have a copy that I could import into Slicer that contained only the bits I need.  
The same job could be done in pretty much any decent CAD program.  Only reason I'd use Blender is I'm most familiar with it, its free, available for all platforms and will run on most machines.  It does have a bit of a learning curve though so something else like meshmixer or ? might be better.   

I guess the point is that the job is more suited to a cad package rather than a slicer.  I suspect that it will be quicker to install and learn the basics of some CAD than for the changes required to be coded, tested and released in Slicer.  I know the Dev's here are good but there's over 1000 open issues and it is possible to already do what you want, even though its tedious to do.

johnbiggs, 2020-04-23:
Really would like an override to let objects float. I have come across this use case before. And it seems that would also solve this request.

rotellavalentino, 2021-02-23:
To split an object: 
          1 import model
          2 select the object you want to cut
          3 push de cut botton regulate the cut and confirm

The object are splitted, but one part overwrite the other.

To solve this:
           1 In up-right you can click on tab advanced
           2 In the list below you have the name of your model splitted in two part
           3 You select one of this two part with a left mouse click
           4 Now in the bad you can see the part you have selected and you can move this one just with mouse

To delete one of this two part: 
            1 In up-right you can click on tab advanced
            2 In the list below you have the name of your model splitted in two part
            3 You select one of this two part with a left mouse click
            4 Now in the bad you can see the part you have selected and you can press 'CANC' botton on the keyboard to delete

The difference between previously version of Prusa Slicer is: When you cut a model, the splitted part are placed on the bad with a distance and you can directly select one of this two part with a mouse, and if you want to delete you can click on CANC botton.

closed 2026-09-29 by prusaslicer-bot[bot]