prusa3d/PrusaSlicer #3975: Ejecting of device has failed on Ubuntu Linux

MarcoAlici, 2020-04-03:
### Version
Version:   2.2.0+linux-x64
Build:     PrusaSlicer-2.2.0+linux-x64-202003211856

### Operating system type + version
Ubuntu 18.04.4 LTS
System Version:      Linux 4.15.0-91-generic x86_64

### 3D printer brand / version + firmware version (if known)
Prusa i3 Plus 1.75 mm
 
### Behavior
The new ejecting button doesn't work. When pressing it, the sd card (micro-sd inserted into its slot on Intel-NUC pc) is not unmounted. The error message: _Ejecting of device [...] has failed._ is showed.

#### Project File (.3MF) where problem occurs
any.

bubnikv, 2020-04-03:
I suppose the eject failed for some reason. Likely the device is in use.

MarcoAlici, 2020-04-05:
> I suppose the eject failed for some reason. Likely the device is in use. 

Me too, I'm here to find the reason. :)
I just insert the card and exported the gcode into it, I can't say who is using it.
I also tryed to open a terminal and `sync` after exporting the gcode, but it doesn't work either.
The usual unmount procedure in Nautilus works.

sebastianha, 2020-04-05:
Same here with when mounting with KDE. Cannot unmount the device from PrusaSlicer.

bear454, 2020-04-15:
Same, GNOME on openSUSE Tumbleweed.

JacekJagosz, 2020-09-17:
Same, Budgie (Gnome based) on Solus Budgie. With both appimage and build manually.

Luxtech, 2020-10-14:
Same with Fedora 32, prusa-slicer installed from fedora repository. ejecting from file manager works just fine.

kocikdav, 2020-11-19:
Hello. Unfortunately we were not able to find computer that would reproduce this issue. I have added some debug logging, maybe it will help us. Please try this build with --loglevel=5.
OSX: https://www.dropbox.com/s/hstwticcsjp1dqk/PrusaSlicer-2.3.0-alpha3%2B96-dk-eject-asan-ga4b9801b8-202011190849.dmg?dl=0
LINUX: https://www.dropbox.com/s/bkiqqlj3oub0xu7/PrusaSlicer-2.3.0-alpha3%2B96-dk-eject-linux-x64-asan-ga4b9801-202011190849.AppImage?dl=0
WINDOWS: https://www.dropbox.com/s/2t5dvuztl6f4yfn/PrusaSlicer-2.3.0-alpha3%2B96-dk-eject-win64-ga4b9801b8-202011190849.zip?dl=0

MarcoAlici, 2020-11-19:
>  I have added some debug logging, maybe it will help us. Please try this build with --loglevel=5.

The Linux version works! The microsd card is correctly unmounted.
Here is the log (hope it helps):
[prusalog.log](https://github.com/prusa3d/PrusaSlicer/files/5568843/prusalog.log)

I also tried with an usb pendrive. It is unmounted, even if the arrow symbol in Nautilus still remains (with the sd card it is correctly deleted after unmounting).

Thank you.

M.

bubnikv, 2021-01-07:
Quite likely the issue was caused by superfluous escaping of spaces in the mount point path. Namely, spaces in the label of the SD card or flash drive would lead to such an issue.

Fixed with https://github.com/prusa3d/PrusaSlicer/commit/2ad4447035548378860ad45d4c649ce5ad06ad50
Closing.

closed 2021-01-07 by bubnikv

sarvenn, 2021-12-31:
@bubnikv hi,
Can't you fix the issue above with another pull request, just like you solved for Linux?