prusa3d/PrusaSlicer #3976: Slicing of part results in PrusaSlicer crash

CraignRush, 2020-04-03:
### Version
__Version:__  2.2.0+win64
__Build:__     PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
**Operating System:**    Windows
**System Architecture:** 64 bit
**Windows Version:**     Windows 10 (build 18363), 64-bit edition
**Total RAM size [MB]:** 12,805MB
#### OpenGL installation
**GL version:**   4.5.0 NVIDIA 376.54
**Vendor:**       NVIDIA Corporation
**Renderer:**     GeForce GT 740M/PCIe/SSE2
**GLSL version:** 4.50 NVIDIA

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S
Firmware: 3.8.1-2869 

### Behavior
* _Describe the problem_
After loading the attached .stl File and positioning it upside down the Slicing process crashes the PrusaSlicer
* _Steps needed to reproduce the problem_
   - Import the .stl
   - Select _Place on Face_
   - Select the top most face (with the engraving)
   - (The slicer is now stuck at Infill Preparation)

      --> The slicer is either in an infinite loop and crashes if you click abort or crashes directly.

* _Actual Results_
  There is no difference if you en- or disable *Background Processing*
[image]

#### Project File (.3MF) where problem occurs
*Here is my project file:*
[Crash.zip](https://github.com/prusa3d/PrusaSlicer/files/4426452/Crash.zip)

neophyl, 2020-04-03:
Slices in about 10 seconds fine on my crappy pc at work.  Windows 10 64 bit, 8gb of ram, 3ghz processor.  Its running on the built in Intel HD 4600 graphics too which arent great.  Open Gl is 4.3.

I suspect its the nvidia graphics as that's had a lot of issues in the past.

CraignRush, 2020-04-03:
Thanks for your fast reply @neophyl! The NVIDIA graphics have been the issue. Running the slicer with the onboard graphics works just fine!

closed 2020-04-03 by CraignRush