prusa3d/PrusaSlicer #3977: UI Compact mode

sugar012, 2020-04-03:
The "issue" is really simple, what about moving the configuration tab to the right and have the plate preview always in sight? I think it could be really helpfull for small adjustements without switching tab everytime. Of course with background processing enabled.
The compact mode it can be enabled in preferences for monitors that has atleast 1920x1080 resolution for greater usability

This mockup is really quick and dirty but you get the idea
[image]

bubnikv, 2020-04-03:
Like this?

[image]

sugar012, 2020-04-03:
The search bar seems a usefull addition and even the toggle, nonetheless. Anyway i find that having all the settings by hand without leaving the plate view can be more usable

thettasch, 2020-04-03:
The "Shortcuts" Tab in the Print Preferences tab in Slic3r allowed you to add some configurations in the plate view. This would be exactly what you are looking for. I used it to quickly change infill pattern, layer height, etc. without having to go to the other tab. It's the feature I miss most from the old Slic3r and would love to see in PrusaSlicer.

bubnikv, 2020-10-16:
We now have an option of floating presets dialog in PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-16 by bubnikv