prusa3d/PrusaSlicer #3981: Unable to use PLA for supports on a PETG project with Wipe Tower enabled

mgg4, 2020-04-04:
### Version
2.2.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3/MMU2 (stock)

### Behavior
* _Describe the problem_
Unable to select a non-soluble filament type to use for support material and use a purge tower.  Trying to use PLA for supports on PETG.  Wipe tower is needed, but I'm not able to specify the extruder for supports and support interfaces *and* enable the purge tower.
* _Steps needed to reproduce the problem_
1. Set the extruder for support and support interface (dismiss the warning popups).
2. Attempt to enable Wipe tower.  Error is presented.
* _Expected Results_
I expected to be able to enable a wipe tower for performing the filament changes necessary for using a PLA material for supports on a PETG object print.
* _Actual Results_
Not able to complete the desired operation.

[image]

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
[stacked visors_Three.zip](https://github.com/prusa3d/PrusaSlicer/files/4430805/stacked.visors_Three.zip)

Arlandw, 2020-07-10:
I have the same issue. Using P2PP with Duet.

rtyr, 2021-02-12:
It should work if you set the contact distance to 0, which you should do anyway, since the PLA and PETG doesn't stick together too much.

bubnikv, 2021-06-03:
The limitation is there so that the support base is printed with an active extruder to avoid unnecessary filament swaps.
Why can't you print the support base with an active extruder?

closed 2026-09-29 by prusaslicer-bot[bot]