prusa3d/PrusaSlicer #3982: Detached Print Settings not displayed

PiteousHonking, 2020-04-04:
### Version
_Version of PrusaSlicer used goes here_
2.2.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Win 10 x64
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
Detached Print Settings are not visible when a user defined printer is selected

* _Steps needed to reproduce the problem_
Select Original Prusa i3 MK3 Printer
Go to Dependancies Tab of a Print Settings Profile
Detach from system preset
Accept default naming with - Detached appended to it
Select new Print Settings
Dependancies Tab - Compatible Printers "All" is selected
Change Printer Settings to a user defined printer
Detached print settings do not appear. 

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-04-04:
It is most likely caused by "compatible printers condition", you have to delete it or update it to match your printer profile. Does it help, can I close this issue?

[image]

PiteousHonking, 2020-04-05:
Edited each Print Setting Profile to delete ~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK(2.5|3).*/ and profiles now visible under user presets. 

Is this the design intention? 
What is the purpose of the Compatible Printers setting if all Print Profiles and all filament profiles have an overriding setting of *PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK(2.5|3)?

PiteousHonking, 2020-04-05:
There is a similar issue with Filament profiles, should I log a separate issue?

rtyr, 2020-04-05:
It is not exactly an issue, PrusaSlicer only respects compatible printers condition as expected.

No, there is no need to open another issue like this. It was already discussed in other issues, for example https://github.com/prusa3d/PrusaSlicer/issues/3729, https://github.com/prusa3d/PrusaSlicer/issues/1322, https://github.com/prusa3d/PrusaSlicer/issues/3015. We might remove whole "compatible printers" feature in the future to make it more unambiguous. Closing.

closed 2020-04-05 by rtyr