prusa3d/PrusaSlicer #3983: After new update, it crashed when I change infill to 100%

ssh003, 2020-04-04:
### Version
PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Windows 10 64bit build 18362

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
When I change Infill to 100%, It crashed. It was okay for the previous version but it happened after I updated to new version for couple weeks ago

rtyr, 2020-04-04:
I am unable to reproduce it. Can you share your 3MF project file? Does it happen everytime?

lukasmatena, 2020-04-04:
If I may, I also have follow-up questions:

1. When it crashes, does it report some error? If you run the application from command line, does it log something?
2. What language are you using? Would it also crash if you switched to English?

Thanks.

ssh003, 2020-04-07:
> I am unable to reproduce it. Can you share your 3MF project file? Does it happen everytime?

Yes it happen everytime and every files after update.
Now I installed previous version so there's no problem but probably I'll get this issue when I upgrade again

ssh003, 2020-04-07:
> If I may, I also have follow-up questions:
> 
> 1. When it crashes, does it report some error? If you run the application from command line, does it log something?
> 2. What language are you using? Would it also crash if you switched to English?
> 
> Thanks.

https://puu.sh/Ft6me.png
https://puu.sh/Ft6m5.png

1. This is the error what I've got it.
I saw exactly same long in here
https://github.com/prusa3d/PrusaSlicer/issues/2279

2. I'm using Korean but even other language got the same issue. I already tried to change English. Although I re-installed PrusaSlicer which is the same version with deleting all of config and profile, there was the issue. BTW I haven't try it on other desktops

Simply I installed previous version, so there's no problem in this time. I don't know it because of my system or it's own problem.

lukasmatena, 2020-04-07:
Thanks. I see the problem in Korean translation, exactly the same that caused #2279 with some of the other translations. Korean was finalized by @ulsanether after #2279 was fixed. That is how it made it into final release. The bug is the single percent sign in translation of
```
#: src/slic3r/GUI/ConfigManipulation.cpp:200
msgid "The %1% infill pattern is not supposed to work at 100%% density."
msgstr "%1% 채우기 패턴은 100% 밀도로 작동하도록 되어 있지 않습니다."
```

> even other language got the same issue. I already tried to change English.

I don't believe that. Can you confirm that the application indeed crashes the same way when it's running in English? It doesn't make sense.

ssh003, 2020-04-08:
> Thanks. I see the problem in Korean translation, exactly the same that caused #2279 with some of the other translations. Korean was finalized by @ulsanether after #2279 was fixed. That is how it [...]

I double checked it. Yes It happened on Korean. May my memory was wrong. I though I'd changed to English to take a picture. BTW I can't use Korean before futher update. So sad :(

lukasmatena, 2020-04-08:
@ssh003 I hope I fixed the problem with ba08d5d.

> I can't use Korean before futher update. So sad :(

That would indeed be sad. I attach the updated PO and MO files. If you transplant them in place of the old ones ((PrusaSlicer folder)/resources/localization/ko), you should be happy again. Could you please confirm the issue disappears if you do that? Thanks.
[ko.zip](https://github.com/prusa3d/PrusaSlicer/files/4449292/ko.zip)

ssh003, 2020-04-09:
> @ssh003 I hope I fixed the problem with [ba08d5d](https://github.com/prusa3d/PrusaSlicer/commit/ba08d5da9e441663f71c7914ef14d830c9e8cbd4).
>
>
> That would indeed be sad. I attach the updated PO and MO [...]

Checked. It works well. Thank you!

closed 2020-04-09 by ssh003